husky/husky #5: Pull args from _description and _gazebo into header.xacro

paulbovbel, 2015-03-25:
Pull args repeated in so many files:

  <xacro:arg name="laser_enabled" default="false" />
  <xacro:arg name="ur5_enabled" default="false" />
  <xacro:arg name="kinect_enabled" default="false" />

Into a single xacro 'header' file.

TheDash, 2015-03-25:
My proposed fix to end all of this nonsense:

Add a $(rosparam ur5_enabled) which searches the rosparam server for that variable.  In the launch file, the params can be set once for the robot and used once in the xacro.

Ofc a pull request will have to be made to xacro

paulbovbel, 2015-03-25:
I'm fairly sure that the launch file params aren't deterministically loaded - there's no guarantee that by the time you load 'robot_description' via xacro calll, that the 'ur5_enabled' param has been stored in the parameter server.

TheDash, 2015-03-25:
True, we would have to solve that as well.

TheDash, 2015-03-25:
It's either we solve that, or have to do the exact same thing for the Jackal+UR3/MICO, the grizzly+UR10, and the LIRP.

Really just a choice of being smart or working hard IMHO.

mikepurvis, 2015-03-25:
Not convinced this medicine is better than the disease. What about env
vars, like Jackal?

TheDash, 2015-03-26:
envvars requires setting params on a bashrc. This is why I believe it is problematic: Why use a bashrc and not something more generic, like rosparam? For each system designed, there will have to be an install sequence that modifies the bashrc for each platform it goes onto. 

In addition, for multi-robot/multi-master/multi-machine, these parameters can't be queried over a network. So launching nodes on two different machines would require having the same param config on each machine. 

The bashrc stuff cannot be dynamically modified in a way that it looks neat. Everything on the systems we're using has access to the param server, whereas only C++/programming languages/scripts can modify the bashrc.

The costs I forsee of implementing a rosparam server to give us these params are:
same usage as optev, but replace optenv with $(rosparam param-name default-value).  

Paul brought up an issue of ensuring that the params are set before their usage, I'm not a multi-threading/process engineer so I can't comment on that, but I understand enough to see that if machine A brings up a node and modifies a rosparam, and machine B brings up the same file but modifies the same rosparam, A might receive B's value. Which, I don't know how rosparam serv handles that but would be worth looking into.

mikepurvis, 2015-03-26:
Let's discuss the various pros and cons further at the meeting next week.

paulbovbel, 2015-06-08:
arg processed before include in xacro, not feasible

closed 2015-06-08 by paulbovbel