husky/husky #17: Error When running movebase with Husky

ktiwari9, 2015-08-19:
Hi there,
I am writing my own package to spawn a husky and make it do a random walk. I obtained the random walk code from https://raw.githubusercontent.com/ras-sight/hratc2015_entry_template/master/examples/cpp/random_walk_navigation.cpp

When I run the code, I get the following error:
Msg Connected to gazebo master @ http://127.0.0.1:11345
Msg Publicized address: 192.168.100.104
[ INFO] [1439967219.405517496]: Laser Plugin (robotNamespace = /), Info: Using the 'robotNamespace' param: '/'
[ INFO] [1439967219.405609058]: Starting Laser Plugin (ns = /)!
[INFO] [WallTime: 1439967219.407472] [0.000000] Spawn status: SpawnModel: Successfully spawned model
[ INFO] [1439967219.407954641]: Laser Plugin (ns = /)  <tf_prefix_>, set to ""
Dbg Plugin model name: -x
[ INFO] [1439967219.438324459, 0.023000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1439967219.473397073, 0.056000000]: Physics dynamic reconfigure ready.
[spawn_husky_model-4] process has finished cleanly
log file: /home/toshiba/.ros/log/2802f36c-460c-11e5-827c-587f66e212a4/spawn_husky_model-4*.log
[ INFO] [1439967220.436458703, 1.006000000]: Waiting for the move_base action server to come online...
[ WARN] [1439967224.459399804, 5.023000000]: Waiting on transform from base_link to map to become available before running costmap, tf error: 

What am I missing here ? All my .yaml files are from the husky_navigation package as it is with no changes and so my husky_urdf where the only thing I changed is adding a hokuyo description to its urdf.

Please advise how to fix this error to simulate a random walk for a husky.

ktiwari9, 2015-08-19:
I am using ROS Hydro with Ubuntu 12.04 Precise and Gazebo 1.9

paulbovbel, 2015-08-20:
The husky stack underwent significant rework for the indigo release, and the navigation demos were written with the new structure in mind. I would recommend running in the simulator in ROS Indigo from debians.

ktiwari9, 2015-08-20:
I managed to fix that bug but I have one more bug now. Husky spawns in Gazebo but it just stays frozen. It doesnt move when I launch move base. I wrote the code following the link https://github.com/ras-sight/hratc2015_entry_template/blob/master/examples/cpp/random_walk_navigation.cpp
when I do rostopic echo /husky/cmd_vel I see no output. Any suggestions on how I can fix this ? My project requires me to stick to ROS Hydro so I cannot upgrade the version. Perhaps with some help with move_base I can make things work in ROS Hydro itself.

paulbovbel, 2015-08-20:
You should investigate where move_base is outputting it's cmd_vel topic.

ktiwari9, 2015-08-21:
Yeah you were right. I just looked at that code again and there is no function to publish velocities to cmd_vel topic. I am trying to visualize a Ackermann Vehicle from https://github.com/wunderkammer-laboratory/ackermann_vehicle/blob/master/ackermann_vehicle_gazebo/launch/ackermann_vehicle.launch to show up beside my husky on the same Gazebo world, but everytime I can only see my husky but not the other robot. My launch file is hosted at http://pastebin.ca/3118965. Is there something I need to change to see both the robots together ?

ktiwari9, 2015-08-21:
So when I can  goal position using a Move_Base setup from your repo how can I make Husky go to that place while avoiding the obstacles using a hokuyo laser scanner ?

paulbovbel, 2015-11-18:
The navigation demos in husky_navigation work out of the box, so you should investigate how those are hooked up if you're trying to implement something on your own.

closed 2016-12-06 by paulbovbel