husky/husky #18: Husky Controller in Gazebo not working

weiweikong, 2015-10-12:
### Setup
- Ubuntu 14.04
- ROS Indigo

### Cannot control Husky througth `cmd_vel` by `teleop_twist_keyboard`
- I would like to import Husky model into a Pixhawk controller-based((https://github.com/PX4/Firmware/blob/master/launch/gazebo_ardrone_house_world.launch) world to do some research about UAV-UGB cooperation. 

- I add the `husky_description` and `husky_controller` into the `rotors_simulator` folder and run the launch file.
```
<launch>
  <arg name="enable_logging" default="false"/>
  <arg name="enable_ground_truth" default="true"/>
  <arg name="log_file" default="ardrone"/>
  <arg name="headless" default="false"/>
  <arg name="gui" default="true"/>

  <env name="GAZEBO_MODEL_PATH" value="$(find rotors_gazebo)/models"/>
  <include file="$(find gazebo_ros)/launch/empty_world.launch">
    <arg name="world_name" value="$(find rotors_gazebo)/worlds/house.world"/>
    <!-- <arg name="debug" value="true" /> -->
    <arg name="headless" value="$(arg headless)"/>
    <arg name="gui" value="$(arg gui)"/>
  </include>

  <group ns="ardrone">
    <include file="$(find rotors_gazebo)/launch/spawn_ardrone.launch">
      <arg name="enable_logging" value="$(arg enable_logging)" />
      <arg name="enable_ground_truth" value="$(arg enable_ground_truth)" />
      <arg name="log_file" value="$(arg log_file)"/>
    </include>
  </group>

  <arg name="laser_enabled" default="false"/>
  <arg name="ur5_enabled" default="false"/>
  <arg name="kinect_enabled" default="false"/>

  <include file="$(find rotors_gazebo)/launch/spawn_husky.launch">
    <arg name="laser_enabled" value="$(arg laser_enabled)"/>
    <arg name="ur5_enabled" value="$(arg ur5_enabled)"/>
    <arg name="kinect_enabled" value="$(arg kinect_enabled)"/>
  </include>

  <!--node name="gazebo_pose_publisher" pkg="rotors_gazebo" type="gazebo_pose_publisher" args="ardrone /ardrone/pose"/-->
</launch>
```
- Both ardrone and Husky could import the Gazebo correctly, however, only ardrone could control by mavros package. The Husky model has no response after sending `cmd_vel` through `teleop_twist_keyboard` package. There was a warning at terminal
[image]

```
process[ekf_localization-7]: started with pid [3286]
process[twist_marker_server-8]: started with pid [3382]
[ INFO] [1444662155.685565964]: [twist_marker_server] Initialized.
process[twist_mux-9]: started with pid [3446]
spawn_model script started
process[spawn_husky_model-10]: started with pid [3503]
process[ardrone/joy_node-11]: started with pid [3508]
process[ardrone/manual_input-12]: started with pid [3569]
[INFO] [WallTime: 1444662155.966920] [0.000000] Loading model xml from ros parameter
[INFO] [WallTime: 1444662155.973443] [0.000000] Waiting for service /gazebo/spawn_urdf_model
process[ardrone/commander-13]: started with pid [3626]
process[ardrone/mc_mixer-14]: started with pid [3685]
process[ardrone/attitude_estimator-15]: started with pid [3812]
[INFO] [WallTime: 1444662156.277059] [0.000000] Calling service /gazebo/spawn_urdf_model
process[ardrone/position_estimator-16]: started with pid [3845]
process[ardrone/mc_att_control-17]: started with pid [3884]
[ INFO] [1444662156.491325069, 0.030000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
process[ardrone/mc_pos_control-18]: started with pid [3973]
process[ardrone/mavlink-19]: started with pid [4069]
[INFO] [WallTime: 1444662156.725853] [0.210000] Spawn status: SpawnModel: Successfully spawned model
[ INFO] [1444662157.284540944, 0.640000000]: Physics dynamic reconfigure ready.
[ardrone/spawn_ardrone-4] process has finished cleanly
log file: /home/amax/.ros/log/44259bf6-70f2-11e5-bcdf-002590fdcfd6/ardrone-spawn_ardrone-4*.log
[spawn_husky_model-10] process has finished cleanly
log file: /home/amax/.ros/log/44259bf6-70f2-11e5-bcdf-002590fdcfd6/spawn_husky_model-10*.log
[WARN] [WallTime: 1444662186.163395] [29.510000] Controller Spawner couldn't find the expected controller_manager ROS interface.
```
It seems that the contorller was not working.

- The published topics are shown as below
[image]

paulbovbel, 2015-10-13:
Spawning a Husky should involve loading a gazebo_ros_control plugin into the Husky model. 

`[WARN] [WallTime: 1444662186.163395] [29.510000] Controller Spawner couldn't find the expected controller_manager ROS interface.`

This suggests that that's not happening, and diff_drive_controller can't be loaded into the ros_control plugin. Try installing the packages from debians, and/or making sure that all dependencies are met first.

weiweikong, 2015-10-13:
Thansk for your reply. 
- Previously, I install all the package through `sudo apt-get install ros-indigo-husky-*`, it also failed when I tried to spwan it in Gazebo with Pixhawk.
- The reason I git it independently is that I would like to add a QR code, such as Apriltag, to do some research about the cooperation control between UAV and UGV. So I need to change the Husky `.xacro` files.
- For above requirements, is there any better way to achieve that?

paulbovbel, 2015-10-13:
What about if you run the Husky sim independently via `roslaunch husky_gazebo husky_playpen.launch`?

weiweikong, 2015-10-13:
It works fine no matter with debians or independent. 
```
......

[ INFO] [1444697945.451994067, 1298.680000000]: Loaded gazebo_ros_control.
[ INFO] [1444697945.457149917, 1298.680000000]: imu plugin missing <xyzOffset>, defaults to 0s
[ INFO] [1444697945.457205517, 1298.680000000]: imu plugin missing <rpyOffset>, defaults to 0s
[ INFO] [1444697945.789543512, 1298.840000000]: Controller state will be published at 50Hz.
[ INFO] [1444697945.792558975, 1298.840000000]: Wheel separation will be multiplied by 1.875.
[ INFO] [1444697945.794470093, 1298.840000000]: Wheel radius will be multiplied by 1.
[ INFO] [1444697945.796309159, 1298.850000000]: Velocity commands will be considered old if they are older than 0.25s.
[ INFO] [1444697945.798065672, 1298.850000000]: Base frame_id set to base_link
[ INFO] [1444697945.799872780, 1298.850000000]: Publishing to tf is disabled
[ INFO] [1444697945.826031575, 1298.880000000]: left wheel to origin: 0.256,0.2854, 0.03282
[ INFO] [1444697945.826103849, 1298.880000000]: right wheel to origin: 0.256,-0.2854, 0.03282
[ INFO] [1444697945.826243680, 1298.880000000]: Odometry params : wheel separation 1.07025, wheel radius 0.17775
[ INFO] [1444697945.831546696, 1298.880000000]: Adding left wheel with joint name: front_left_wheel and right wheel with joint name: front_right_wheel
[ INFO] [1444697945.831661836, 1298.880000000]: Adding left wheel with joint name: rear_left_wheel and right wheel with joint name: rear_right_wheel
```
I received the above reply correctly, which I think shoud be load the control package througth `control.launch`. However, as the Husky Spawning to other world, I got no related information.

paulbovbel, 2015-10-13:
This looks like either an issue with your multi-robot namespacing or the way you've setup your custom xacro's. Either way, that's not an issue with the published software. I'd suggest opening a ROS Answers question, and providing a link to your custom code.

weiweikong, 2015-10-13:
Thank you for your replies. I will check the namespacing issue.

closed 2015-10-14 by paulbovbel