husky/husky #25: hsky_ur5_moveit gazebo inertial not stable

andrestoga, 2016-03-31:
Hi,

I am simulating husky_ur5_moveit package in ROS indigo following this [tutorial](http://wiki.ros.org/husky_ur5_moveit_config/Tutorials/Husky%20UR5%20Mobile%20Manipulation%20Demo) but its inertial parameters are not stable.

It appears as [here](http://i65.tinypic.com/25036u8.png).

Is this an old version?

Best,

paulbovbel, 2016-04-01:
FYI @tonybaltovski sounds the Husky intertial params could use some tuning. 

@TheDash, did you see this when you were developing the moveit config/demos?

paulbovbel, 2016-04-04:
One option for moving the laser around is via environment variables (http://wiki.ros.org/husky_bringup/Tutorials/Customize%20Husky%20Configuration#Environment_Variables)

I'll have to leave it with @TheDash to comment on the moveit configuration.

CMobley7, 2016-07-05:
@andrestoga @paulbovbel @TheDash 

The inertial stability remains an issue for me in simulation. I adjusted the decorations.urdf.xacro to remove the front bumper and moved the lidar to its position in husky.urdf.xacro. As shown in the image below. In addition, I increased the husky's mass and proportionally the inertia in each direction. 

[image]

While the robot spawns in the correct position now and stability overall improved, the husky turns right very slowly over time. In addition, any movement of the arm outside the bounds of the top_plate has a tendency to cause a lot of motion of the base. After observing the husky's behavior in simulation, I believe the issue could be caused by the ur5_arm_shoulder_lift_joint. As the majority of the motion occurred when the two link don't appear to be connected properly as in the image below. Any suggestion you can give on fixing this issue would be appreciated. 

[image]

Thank you for your time and attention to this matter. Have a great week. God bless.

Very Respectfully,
CMobley7

TheDash, 2016-07-05:
> The inertial stability remains an issue for me in simulation. I adjusted the decorations.urdf.xacro to remove the front bumper and moved the lidar to its position in husky.urdf.xacro. As shown in [...]

For best stability - I always just used laser_enabled:=false. However, that shouldn't have to be the case.

> > 
> While the robot spawns in the correct position now and stability overall improved, the husky turns right very slowly over time. 

I've noticed this too.

> As the majority of the motion occurred when the two link don't appear to be connected properly as in the image below. Any suggestion you can give on fixing this issue would be appreciated.

I'll check what changing the position of the UR5 does. It looks like it may be too far off of the platform, causing all of this drifting and swaying.

CMobley7, 2016-07-15:
@TheDash,

I measured the physical top_plate on our Husky and its dimensions were .825 m x .620 m x .006 m. However, the STL provided here has the top_plate dimensions as  .73 mm x .59 mm x 0.01 mm. I made a new top_plate with the physical dimensions above and exported it as an STL and DAE. I replaced the top_plate files in husky_description and expected to have to change the import size in the urdf. However, strangely it imported with the correct size. Now, the ur5_arm position match's on the physical and simulation Husky. However, the Husky now moves around erratically regardless of what is enabled. In an empty world, it will spawn as seen in the picture below and move around. I'm going to adjust my part later today from m to mm and see if this has any effect. Any additional ideas? Did changing the position of the ur5_arm in your urdf have any effect on the Husky's motions while the ur5_arm was moving?

[image]

CMobley7, 2016-07-16:
@TheDash,

Adjusting it from m to mm fixed the erratic movement on startup. However, the adjusted plate dimensions had no effect on the stability that I can see. Any suggestions?

Thank you for your time and attention to this matter. Have a great weekend. God bless.

Very Respectfully,
CMobley7

tonybaltovski, 2018-04-12:
Closing to due inactivity, please feel free to reopen if the needed.

closed 2018-04-12 by tonybaltovski