husky/husky #27: Error when planning with Moveit! Husky UR5 arm

andrestoga, 2016-04-05:
Hi,

I followed this [tutorial](http://wiki.ros.org/husky_ur5_moveit_config/Tutorials/Husky%20UR5%20Mobile%20Manipulation%20Demo) to use the Husky with the UR5 using Moveit!. Everything works ok so far. Now I want to use Moveit! to load motion planners at runtime. I created a new package to create a new node that will be in charge of execute the planning. This is my code:

```
#include <ros/ros.h>
#include <moveit/move_group_interface/move_group.h>

int main(int argc, char **argv)
{
  ros::init(argc, argv, "lesson_moveit");
  moveit::planning_interface::MoveGroup group("manipulator");

  // start a background "spinner", so our node can process ROS messages
  //  - this lets us know when the move is completed
  ros::AsyncSpinner spinner(1);
  spinner.start();

  // INSERT MOTION COMMANDS HERE
}
```

I got the following errors when trying to run the node:

```
andrestoga@andrestoga-RoboLab:~/ros_motion_planning$ rosrun project2 ur5_mover 
[ERROR] [1459832675.112257416]: Semantic description is not specified for the same robot as the URDF
[ERROR] [1459832675.112361080]: Link 'base_laser' is not known to URDF. Cannot disable collisons.
[ERROR] [1459832675.112377176]: Link 'base_laser' is not known to URDF. Cannot disable collisons.
[ERROR] [1459832675.112388309]: Link 'base_laser' is not known to URDF. Cannot disable collisons.
[ERROR] [1459832675.112399055]: Link 'base_laser' is not known to URDF. Cannot disable collisons.
[ERROR] [1459832675.112419331]: Link 'base_laser' is not known to URDF. Cannot disable collisons.
[ERROR] [1459832675.112430294]: Link 'base_laser' is not known to URDF. Cannot disable collisons.
[ INFO] [1459832675.112529966]: Loading robot model 'husky_robot_gazebo'...
[ INFO] [1459832675.112553399]: No root joint specified. Assuming fixed joint
[ERROR] [1459832675.279646657]: Error retrieving file [file:///opt/ros/indigo/share/ur_description/meshes/ur5/collision/upperarm.stl]: Couldn't open file /opt/ros/indigo/share/ur_description/meshes/ur5/collision/upperarm.stl
[ERROR] [1459832675.297248854]: Error retrieving file [file:///opt/ros/indigo/share/ur_description/meshes/ur5/visual/upperarm.dae]: Couldn't open file /opt/ros/indigo/share/ur_description/meshes/ur5/visual/upperarm.dae
[ERROR] [1459832675.333689794, 4214.540000000]: Error retrieving file [file:///opt/ros/indigo/share/ur_description/meshes/ur5/collision/wrist1.stl]: Couldn't open file /opt/ros/indigo/share/ur_description/meshes/ur5/collision/wrist1.stl
[ERROR] [1459832675.352395002, 4214.560000000]: Error retrieving file [file:///opt/ros/indigo/share/ur_description/meshes/ur5/visual/wrist1.dae]: Couldn't open file /opt/ros/indigo/share/ur_description/meshes/ur5/visual/wrist1.dae
[ERROR] [1459832675.370355017, 4214.570000000]: Error retrieving file [file:///opt/ros/indigo/share/ur_description/meshes/ur5/collision/wrist2.stl]: Couldn't open file /opt/ros/indigo/share/ur_description/meshes/ur5/collision/wrist2.stl
[ERROR] [1459832675.388504851, 4214.590000000]: Error retrieving file [file:///opt/ros/indigo/share/ur_description/meshes/ur5/visual/wrist2.dae]: Couldn't open file /opt/ros/indigo/share/ur_description/meshes/ur5/visual/wrist2.dae
[ERROR] [1459832675.406530003, 4214.610000000]: Error retrieving file [file:///opt/ros/indigo/share/ur_description/meshes/ur5/collision/wrist3.stl]: Couldn't open file /opt/ros/indigo/share/ur_description/meshes/ur5/collision/wrist3.stl
[ERROR] [1459832675.424788651, 4214.630000000]: Error retrieving file [file:///opt/ros/indigo/share/ur_description/meshes/ur5/visual/wrist3.dae]: Couldn't open file /opt/ros/indigo/share/ur_description/meshes/ur5/visual/wrist3.dae
[ERROR] [1459832675.499274924, 4214.700000000]: The kinematics plugin (ur5_arm) failed to load. Error: According to the loaded plugin descriptions the class ur_kinematics/UR5KinematicsPlugin with base class type kinematics::KinematicsBase does not exist. Declared types are  kdl_kinematics_plugin/KDLKinematicsPlugin srv_kinematics_plugin/SrvKinematicsPlugin
[ERROR] [1459832675.499357300, 4214.700000000]: Kinematics solver could not be instantiated for joint group ur5_arm.
[FATAL] [1459832675.499436556, 4214.700000000]: Group 'manipulator' was not found.
terminate called after throwing an instance of 'std::runtime_error'
  what():  Group 'manipulator' was not found.
Aborted (core dumped)
```

I don't know why it couldn't open the files and load the kinematics plugin for the ur5 arm. Any advice in how to resolve it will be helpful.

Best,

TheDash, 2016-05-10:
Hi there,

The collisions need to be updated, I'll do that later this afternoon. In addition, I added a dependency on ur_kinematics, which you were probably missing to use the plugin

CMobley7, 2016-07-01:
@andrestoga,

Was this issue resolved? Where you successful in getting the ur5_arm to go to a specified point?

TheDash, 2016-07-01:
@CMobley7 

Do you have ur_kinematics installed either via source or via deb?

CMobley7, 2016-07-01:
@TheDash,

I installed ur_kinematics from source.

CMobley7, 2016-07-01:
@andrestoga 

If you are still having this problem, I suggest changing your solver to trac_ik. I just install trac-ik-kinematics-plugin and changed my kinematics.yaml file from

```
ur5_arm:
  kinematics_solver: ur_kinematics/UR5KinematicsPlugin
  kinematics_solver_search_resolution: 0.005
  kinematics_solver_timeout: 0.005
  kinematics_solver_attempts: 3
```

to

```
ur5_arm:
  kinematics_solver: trac_ik_kinematics_plugin/TRAC_IKKinematicsPlugin
  kinematics_solver_timeout: 0.05
  solve_type: Manipulation1
```

It has successfully solved every IK problem I have given it.

closed 2016-11-21 by TheDash

Raffa87, 2017-02-09:
Hi, 
I am getting the same problem, but change of kinematic solver doesn't help me :(

CMobley7, 2017-02-09:
Is this in simulation or on a physical husky?

Raffa87, 2017-02-10:
On a physical husky

CMobley7, 2017-02-13:
The husky_ur5_planning_execution.launch file I used was slightly different than the default.

```
<launch>
  <arg name="real" default="true" />
  <arg name="limited" default="true"/>
  <arg name="debug" default="false" />

  <!-- Remap follow_joint_trajectory -->
  <remap if="$(arg real)" from="arm_controller/follow_joint_trajectory" to="follow_joint_trajectory"/>

  <!-- Launch moveit -->
  <include file="$(find husky_ur5_moveit_config)/launch/move_group.launch">
    <arg name="debug" default="$(arg debug)" />
  </include>
</launch>
```

Also, are you running all of the following commands?

```
roslaunch ur5.launch
roslaunch husky_ur5_moveit_config husky_ur5_planning_execution.launch limited:=true real:=true
roslaunch husky_ur5_moveit_config moveit_rviz.launch config:=true
```