husky/husky #31: No error but the robot just do not move.

YajingWang1994, 2016-05-25:
Hi,
 I took your husky packages (husky_description, husky_gazbeo, and husky_control) as reference to build my working packages, but  use my own model and world. 

I could get command news from my velocity_controller (which should be directly subscribed by gazebo), but my mobile robot just do not move and actually no reaction. 

There were no errors reported, so I could not find out why it does not work.

I hope to get some instructions here.

Or, Is there any way that I can check the news received by gazebo?

Looking for your answers.
I would be really grateful.

paulbovbel, 2016-05-25:
Please post to ROS answers, this does not sound like a husky-specific issue.

CMobley7, 2016-07-01:
@YajingWang1994,

I did something similar in https://github.com/CMobley7/team_7_ws/tree/master/src/truck. Maybe looking through that code with help solve your issue if it still remain. If not please close this issue.

Best,
CMobley7

tonybaltovski, 2016-10-03:
Due to inactivity.

closed 2016-10-03 by tonybaltovski