husky/husky #37: Moving UR5

paulbovbel, 2016-11-29:
As an effort to release husky for kinetic, I am planning on moving all UR5 related bringup to a separate package to break the dependency on UR5 and moveit.

This will be a good chance to get a stable config for single/dual UR5 husky into a main branch. The problem will be with any outstanding PRs. I will do my best to blitz through anything outstanding, and keep anything that looks sane.

FYI @tonybaltovski @TheDash @andrestoga @wxmerkt @k-okada

paulbovbel, 2016-11-29:
ETA Dec 23rd

wxmerkt, 2016-11-29:
@paulbovbel Thanks for the heads-up! Is there any plan for splitting/different default configs for localization.yaml, planning.yaml etc for different configs? Apart from the gear box ratio, we've noticed that the values for ros-navigation need to be changed to be more conservative. We'll likely put in another PR soon to fix the IMU frame optenv on the dual-arm-husky.

@TheDash Are there plans to unify the dual-arm xacros with above ETA?

TheDash, 2016-11-29:
To be honest. I've always thought about the design like this:

1. The husky is a product
2. The husky UR5/dual UR5 is a product

Rather than this:

1. The husky is a product that maintains a sub integration of a UR5/dual UR5

Rationale: there are simply too many differences between the each to logically maintain them in the same piece of code. Rather, I'd like it more modular. Future Husky changes can be backported to the dual UR5 branches to get access to new features.

As such, there should be separate moveit packages that are installed in a for-use basis for shipping the robots, and the code for the husky will change on a per branch basis. E.g installing the husky code by source and overlaying it with the husky_dual_ur5 branch.

TheDash, 2016-11-29:
@paulbovbel @tonybaltovski how do you feel about doing things this way? 

This is a unique situation since we sell UR5 Husky and Dual UR5 husky as separate products, rather than a base husky + accessories which should be in one package. In addition, if there are separate branches, it is known then that I will care/work with the PR's/maintenance for those branches, and maintainership is well defined.

The user and customers will also know that, if they are using the dual_ur5 branch that everything is contained within that branch, and all the parameters/tweaks are for their robot, not to be confused wondering if the main husky branch should work for their husky.

If @paulbovbel and @tonybaltovski agree with this, I'll go ahead and close the PR #33  and @tonybaltovski @jeff-o myself will then focus on  making sure one product works at a time, rather than maintaining all with one PR.

jeff-o, 2016-11-29:
I’m all for separating and tuning separate branches for single and dual arm Huskies.

paulbovbel, 2016-11-29:
We can't keep mainline software in branches. Anything stable and 'standard' needs to be in a package and released as a debian.

Any customization needs to have a sane workflow. This will likely be by via passing around customizations via launch-args + env-vars.

The xacro customizations, I'm looking to standardize on one scheme with Jackal, so probably URDF_EXTRAS.

paulbovbel, 2016-11-29:
husky_ur5 and husky_dual_ur5 can be first class customization under the scheme. moveit_configs need to exist for those as well, so we'll have to touch base on how those are generated later.

TheDash, 2016-11-29:
The customization workflow is aside to what this proposal is trying to achieve. And I do agree its a mess. The software should contain everything necessary to configure the robot, including the docs/manuals that list and explain the parameters to be used in the robot configurations.

I agree that anything stable and standard needs to be packaged, but we can also bloom the ur5 branches separately.

paulbovbel, 2016-11-29:
You can't bloom separate branches in one rosdistro like that.

Anyways, my primary motivation here is to get extraneous dependencies (i.e. moveit and ur5) out of the primary husky packages. Pushing UR5 husky config into a repo that can be used as a 'customization' of the base husky is a start. I'm relatively open minded to how the customization scheme is laid out, and will play around with it on that day. It'll probably look a lot like what it looks like now, but with more injection points for control and etc.

Considering neither config is stable as is, the cost is minimal. @TheDash, if you like you can put time into stabilising and cleaning up the ur5 story after that.

This will also be kinetic only, meaning UR/moveit will have to wait until the upstream release.

TheDash, 2016-11-29:
Anyways, my primary motivation here is to get extraneous dependencies (i.e. moveit and ur5) out of the primary husky packages.

That means we will have to go the branched route. The urdf.xacro files will always have to include dependencies on the universal_robot packages  as long as it is including its description. In addition, there are other dependencies the UR5 product brings along such as robotiq grippers, force torques, and custom husky hardware design. The best way to do this is branching it out entirely. Branching is an excellent form of customization. 

The problem with doing it as a UR5 customizes a base husky, is that the UR5 config actually modifies the base husky, and the base husky intends to be non modified intrinsically, and those two forces go against each other, creating massive complexity. 

For example, on a dual ur5 config i had to:
1. remove the include of the top plate and parametrize it and make an env var for it
2. remove the default lidar config position and set up new env vars for the two new lidar positions and their parent links
3. remove the user rails
4. add a custom top plate (bulk head) which fits the rig of the arms

When these things modify so much of the husky base, it becomes a new base/standard config itself.

tonybaltovski, 2016-11-29:
I envision that the Husky is a product and the single or dual UR5 configuration is an extension.  The main Husky packages will have to be changed to allow changes to be applied from the ur5 integration packages on to them (such as different `ros_control` parameters that the ur5 integration would need).  Using code on branches should be totally avoided.

TheDash, 2016-11-29:
> I envision that the Husky is a product and the single or dual UR5 configuration is an extension. The main Husky packages will have to be changed to allow changes to be applied from the ur5 [...]

It seems that way. But when you look at how much of the base husky had to be changed so that we could add arms and make a standard configuration (derived from business requirements), then it is clear that it is actually not as much of an extension as it is a type of base platform.

tonybaltovski, 2016-11-29:
> For example, on a dual ur5 config i had to:
> 1. remove the include of the top plate and parametrize it and make an env var for it
> 2. remove the default lidar config position and set up new env vars [...]

1. This is useful to the base husky.
2. Could the parent links not be change by environment? If not, this could be a useful addition.
3. This is useful to the base husky. @jeff-o was even asking for this.
4. This could be contained in a xacro that URDF_EXTRAS will point to.

paulbovbel, 2016-12-01:
+1 to all that @tonybaltovski 

@TheDash those things that had to be done to the general model to support more extensibility, but can be used for all customization.

The key is how to chain the customizations properly via env-vars -> roslaunch args -> xacro, which I think we've gotten better at over time.

paulbovbel, 2017-01-25:
UR5 and moveit related pieces have been removed from kinetic-devel for purposes of release.

urdf_extras and config_extras in launch files are provided as hooks to inject addition robot description, or control parameters. Providing a UR5 customization package on top of the 'base' husky is best-effort at this point.

closed 2017-01-25 by paulbovbel