husky/husky #38: FLIR PTU Bumblebee mounting bracket not part of model

wxmerkt, 2016-12-03:
The mounting bracket connecting the Bumblebee2 to the PTU isn't part of the Dual Arm Husky URDF and the centre of rotation thus offset by the height of the bracket. Could you please add the bracket to the model/repository/URDF?

Thanks

Cf. https://github.com/husky/husky/tree/dual_ur5_husky/husky_description/meshes/accessories and https://github.com/husky/husky/tree/dual_ur5_husky/husky_description/meshes/

wxmerkt, 2016-12-03:
Actually, self-observing with a camera mounted atop the rear end the whole mounting appears to be off, shifted for both the base of the PTU and the bracket for the bumblebee (this is after adding a 2cm offset for the bracket):

[image]

wxmerkt, 2016-12-03:
It's off both in X and Y because the base_link has an offset from the mount_link due to the morphology of the PTU, cf:

[image]

[image]

wxmerkt, 2016-12-03:
I've proposed a fix to this in #39.

tonybaltovski, 2017-05-08:
@wxmerkt @TheDash I am going to close this since it part of the Dual Arm Husky URDF.

closed 2017-05-08 by tonybaltovski