husky/husky #40: UR5 Stow Scripts not working

wxmerkt, 2016-12-06:
Hi @TheDash 
The stow scripts for our UR5s are not working, it waits forever for the server to come up. We are running firmware version ``3.2.18744`` and the same issue has been reported upstream (https://github.com/ThomasTimm/ur_modern_driver/issues/27#issuecomment-220976342).

Have you run the stow script on the robot and do you have any idea why it may be "waiting for server" forever?

For now we are using our own planning and motion execution pipeline to stow the arms to a desired configuration but it would be great to know why it's not working.

Thanks,
Wolfgang

TheDash, 2016-12-06:
It is possible that the drivers for the arm aren't coming up/crashing. What does your rostopic list say? Can you ping the arms? What happens when manually launching the arm drivers in /etc/ros/indigo/husky-core.d?

wxmerkt, 2016-12-06:
The drivers work definitely because we are commanding the arms via a SimpleActionClient at the same time when the stow scripts dont work (just tried again).

Manually launching the arm drivers makes them conflict with the ones loaded on boot and they restart alternating - causing communication delays. We figured out that the best mode to operate the robot is to turn on the husky, turn on the arms, restart the husky-core service. Then we dont need to manually start the arms.

TheDash, 2016-12-06:
Is this an issue with ur_modern_driver then? If it's waiting for server it means that the ur_modern_driver isn't publishing the topic its looking for. Server in this case is referencing the simple action server/trajectory server. If it is indeed up, looks like it might be namespacing issue and a topic relay/remapping will be required.

Can you post your rostopic list and rosservice list for the robot?

wxmerkt, 2016-12-06:
Oh yeah, it was a namespacing issue. It can now connect to the server and says so but doesn't move:

```bash
$ rosrun husky_control stow_ur5_left
Waiting for ur5_arm server...
Connected to ur5_arm server
```

I added debug printing and it apparently receives a result immediately after sending the goal (i.e. it immediately completes wait_for_result without having moved at all

TheDash, 2016-12-06:
Is it in stowed position already? Modifying https://github.com/husky/husky/blob/indigo-devel/husky_control/scripts/stow_ur5#L11 can change its stow position

wxmerkt, 2016-12-06:
Nope, I've moved it into a non-stowed position using our planning/execution pipeline. It still didn't move when commanding to stow it.

TheDash, 2016-12-07:
Sorry, can you post what  command is being run?

should be something like

rosrun husky_control stow_ur5_left

wxmerkt, 2016-12-07:
a) command being run is ``rosrun husky_control stow_ur5_left``
b) I had to change the SimpleAction client to "husky_left_ur5/follow_joint_trajectory" which is the one we are successfully using for executing our trajectories. With the "l_arm_controller" it waits forever for the server to come up.
c) It exits the move() method immediately from the "wait_for_result" line (confirmed by adding a print after wait_for_result

tonybaltovski, 2017-05-08:
@wxmerkt @TheDash I am going to close this since it part of the Dual Arm Husky.

closed 2017-05-08 by tonybaltovski