husky/husky #41: Microstrain Automatic Magnetometer Calibration

wxmerkt, 2016-12-06:
We are seeing a yaw drift from the ROS EKF when stationary. Trying to run the calibration as outlined in the [Husky instructions](https://www.clearpathrobotics.com/assets/guides/husky/InstallHuskySoftware.html#calibrating-the-magnetometer) doesn't work:

```bash
administrator@cpr-ued04:~$ rosrun husky_bringup calibrate_compass
Started rosbag record, duration 60 seconds, pid [23975]
Started motion commands, pid [23976]
[ INFO] [1481042907.095201284]: Subscribing to /imu/data
[ INFO] [1481042907.098019893]: Subscribing to /imu/mag
[ INFO] [1481042907.100714134]: Subscribing to /tf
[ INFO] [1481042907.104380517]: Recording to /tmp/calibrate_compass.WXw7/imu_record.bag.
Test underway.
Time remaining: 0   
Shutting down motion command publisher.
Waiting for rosbag to shut down.
Computing magnetic calibration.
Unable to compute calibration from recorded bag.
Output in /tmp/compute_output.log
administrator@cpr-ued04:~$ cat /tmp/compute_output.log 
Using 0 samples.
Traceback (most recent call last):
  File "/home/administrator/dual_ur5_ws/install/lib/husky_bringup/compute_calibration", line 56, in <module>
    x,y,z = zip(*vecs)
ValueError: need more than 0 values to unpack
administrator@cpr-ued04:~$ 
```

The Husky is rocking while rotating. Any advice on whether this script should work for the Microstrain IMU on the Dual UR5 Husky?

cc @TheDash

paulbovbel, 2016-12-06:
Magnetometer calibration is only used with a UM6/UM7 IMU, due to the software imu_filter_madgwick instance used in the pipeline.

Microstrain provides their own proprietary hard/soft iron calibration tooling.

closed 2016-12-06 by paulbovbel

wxmerkt, 2016-12-06:
Thanks, was wondering whether it was meant to work