husky/husky #49: Invalid parameter "robot_namespace"

nobinov, 2017-05-02:
Hello,

I am willing to run Husky Robot on my Gazebo. However, I'm using Kinetic, and current stable version of Husky in repo is still Indigo. So I tried to cloned this repo to my new catkin workspace, compiled it using `catkin_make` and run `roslaunch husky_gazebo husky_empty_world.launch `.

However, I got this error:
`Invalid parameter "robot_namespace"
XacroException('Invalid parameter "robot_namespace"',)
when instantiating macro: sick_lms1xx (/opt/ros/kinetic/share/lms1xx/urdf/sick_lms1xx.urdf.xacro)
in file: /home/novian/catkin_husky/src/husky/husky_description/urdf/husky.urdf.xacro
while processing /home/novian/catkin_husky/src/husky/husky_gazebo/launch/spawn_husky.launch:
while processing /home/novian/catkin_husky/src/husky/husky_description/launch/description.launch:
Invalid <param> tag: Cannot load command parameter [robot_description]: command [/opt/ros/kinetic/lib/xacro/xacro '/home/novian/catkin_husky/src/husky/husky_description/urdf/husky.urdf.xacro'     --inorder     robot_namespace:=/     laser_enabled:=true     kinect_enabled:=false     urdf_extras:=     ] returned with code [2]. 

Param xml is <param command="$(find xacro)/xacro '$(find husky_description)/urdf/husky.urdf.xacro'     --inorder     robot_namespace:=$(arg robot_namespace)     laser_enabled:=$(arg laser_enabled)     kinect_enabled:=$(arg kinect_enabled)     urdf_extras:=$(arg urdf_extras)     " name="robot_description"/>
The traceback for the exception was written to the log file
`

Is something missing?

Thanks

tonybaltovski, 2017-05-03:
The lms1xx package hasn't been released with the needed changes for this.  If you install it from source it will work.

Let me know how it goes.

nobinov, 2017-05-04:
It works!
(At least until now the Husky spawned and simulator run properly. I still didn't try laser sensor (lms1xx) yet)

Thanks for the advice.

tonybaltovski, 2017-05-04:
Just released 0.1.6 of LMS1xx.  It should be synced soon.

closed 2017-05-04 by tonybaltovski