husky/husky #50: Different odometry/filtered results on different computers

gmsanchez, 2017-05-10:
Hi,

I am simulating Husky on ROS Kinetic on Ubuntu 16.04.2. I was trying to make some controllers in order to find the velocities that drive the Husky to a desired point in space, but I started to get different results on differents computers (my notebook and my desktop). I found out that odometry/filtered change a lot from one computer to the other and I can't figure out why is that happening.

I made a Python script that sends fixed controls to the Husky. While the odometry/filtered is fairly correct on the notebook, on the desktop I get that the real position [x,y,yaw] is [10.254782; 8.853166; 0.748164] while odometry/filtered publishes [7.7668092; 11.26777496; 1.01970051].

You can try to download the script I used to get the results from http://github.com/gmsanchez/husky_mpc and run [apply_fixed_controls.py](https://github.com/gmsanchez/husky_mpc/blob/master/apply_fixed_controls.py).

Thanks in advance, any help will be appreciated.

paulbovbel, 2017-05-10:
This is very much not a husky issue. I can say offhand that the way you setup your experiment will significantly affect your results (since you're feeding velocity commands open-loop to a simulated robot, which will not be deterministic).

If you can attach actual data and document your findings with more detail ('change a lot' is not very valuable to anyone trying to help), you should post them to ROS Answers and see if the greater ROS community can help explain what you're seeing.

It may be an issue with simulation, with sensor processing, or as I mentioned above, with a non-deterministic experimental setup.

closed 2017-05-10 by paulbovbel

gmsanchez, 2017-05-10:
Thanks for the feedback. I know that the description I provided is very vague, but I thought that providing the example where the odometry/filtered doesn't work well was enough. I will gather some more data and post this question to ROS Answers.