husky/husky #66: How to simulate husky in ROS kinetic version

hoppss, 2017-07-17:
I am a new learner to husky . I install the Kinetic and husky pacakge from this git.I meet following errors.Could you help me ?
`roslaunch husky_gazebo husky_empty_world.launch
... logging to /home/robot/.ros/log/82c86086-6a98-11e7-945c-000c29977d69/roslaunch-ubuntu-10295.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/__init__.py", line 307, in main
    p.start()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/parent.py", line 268, in start
    self._start_infrastructure()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/parent.py", line 217, in _start_infrastructure
    self._load_config()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/parent.py", line 132, in _load_config
    roslaunch_strs=self.roslaunch_strs, verbose=self.verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/config.py", line 451, in load_config_default
    loader.load(f, config, verbose=verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 746, in load
    self._load_launch(launch, ros_config, is_core=core, filename=filename, argv=argv, verbose=verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 718, in _load_launch
    self._recurse_load(ros_config, launch.childNodes, self.root_context, None, is_core, verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 682, in _recurse_load
    val = self._include_tag(tag, context, ros_config, default_machine, is_core, verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 95, in call
    return f(*args, **kwds)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 625, in _include_tag
    default_machine, is_core, verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 652, in _recurse_load
    default_machine, is_core, verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 652, in _recurse_load
    default_machine, is_core, verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 682, in _recurse_load
    val = self._include_tag(tag, context, ros_config, default_machine, is_core, verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 95, in call
    return f(*args, **kwds)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 625, in _include_tag
    default_machine, is_core, verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 682, in _recurse_load
    val = self._include_tag(tag, context, ros_config, default_machine, is_core, verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 95, in call
    return f(*args, **kwds)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 587, in _include_tag
    inc_filename = self.resolve_args(tag.attributes['file'].value, context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 183, in resolve_args
    return substitution_args.resolve_args(args, context=context.resolve_dict, resolve_anon=self.resolve_anon)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 370, in resolve_args
    resolved = _resolve_args(resolved, context, resolve_anon, commands)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 383, in _resolve_args
    resolved = commands[command](resolved, a, args, context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 151, in _find
    source_path_to_packages=source_path_to_packages)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 197, in _find_executable
    full_path = _get_executable_path(rp.get_path(args[0]), path)
  File "/usr/lib/python2.7/dist-packages/rospkg/rospack.py", line 203, in get_path
    raise ResourceNotFound(name, ros_paths=self._ros_paths)
ResourceNotFound: multimaster_launch
ROS path [0]=/opt/ros/kinetic/share/ros
ROS path [1]=/home/robot/husky/src
ROS path [2]=/opt/ros/kinetic/share
`
Could you help me  how to run the kinetic husky pacakage ?

WillGrayMSU, 2017-09-20:
The indigo release is more stable, and is what I'd recommend to new learners.  That being said, your initial issue can be resolved by installing the mulitmaster_launch package.

`sudo apt-get install ros-kinetic-multimaster-launch`

Next you will need to install the Kinetic packages for the simulated LIDAR.

`sudo apt get install ros-kinetic-lmslxx`

Beyond this, I am still working things out myself.  If anyone else has advice beyond this, please update us.

ShinjiOhyama, 2017-09-21:
I needed the following packages to make it work in ROS Kinetic.
```
sudo apt-get install ros-kinetic-multimaster-launch
sudo apt-get install ros-kinetic-lms1xx 
sudo apt-get install hector-gazebo
```

Also I installed the below somehow but they might not be needed actually.
```
sudo apt-get install ros-kinetic-interactive-marker-twist-server 
sudo apt-get install ros-kinetic-twist-mux
sudo apt-get install ros-kinetic-imu-tools
```

andrestoga, 2017-11-23:
Hi,

I'm still getting the following errors but at least I can see the model of the Husky in Gazebo.

`spawner: error: unrecognized arguments: -shutdown-timeout 3
[base_controller_spawner-9] process has died [pid 20075, exit code 2, cmd /opt/ros/kinetic/lib/controller_manager/spawner husky_joint_publisher husky_velocity_controller - -shutdown-timeout 3 __name:=base_controller_spawner __log:=/home/andrestoga/.ros/log/f0d7a45e-d09d-11e7-9e74-d8eb97ea52bc/base_controller_spawner-9.log].
log file: /home/andrestoga/.ros/log/f0d7a45e-d09d-11e7-9e74-d8eb97ea52bc/base_controller_spawner-9*.log`

`[FATAL] [1511476245.517167500, 0.166000000]: No matching hardware interface found for 'hardware_interface/VelocityJointInterface
[FATAL] [1511476245.517204447, 0.166000000]: Could not initialize robot simulation interface`

andrestoga, 2017-11-25:
To fix the following error:

`[FATAL] [1511579416.056199650, 0.173000000]: No matching hardware interface found for 'hardware_interface/VelocityJointInterface
[FATAL] [1511579416.056250253, 0.173000000]: Could not initialize robot simulation interface
`
Just replace the lines from 68 to 70 in this file:

https://github.com/husky/husky/blob/kinetic-devel/husky_description/urdf/wheel.urdf.xacro

with the following code:

`<hardwareInterface>VelocityJointInterface</hardwareInterface>
			</actuator>
			<joint name="${wheel_prefix}_wheel">
				<hardwareInterface>VelocityJointInterface</hardwareInterface>`

AdityaPrakash23, 2018-01-03:
Hello There!
I am trying the Husky robot simulation for ROS kinetic. I am following the steps in [http://wiki.ros.org/husky_gazebo/Tutorials/Simulating%20Husky](http://wiki.ros.org/husky_gazebo/Tutorials/Simulating%20Husky). But I am getting this error  
`ERROR: cannot launch node of type [controller_manager/spawner]: controller_manager
ROS path [0]=/opt/ros/kinetic/share/ros
ROS path [1]=/home/aditya/project_ws/src
ROS path [2]=/opt/ros/kinetic/share
process[ekf_localization-10]: started with pid [21396]
process[twist_marker_server-11]: started with pid [21399]
[WARN] [1514969622.115142, 0.000000]: Master_discovery node appear not to running. Wait for topic with type 'MasterState.
process[robot_state_publisher-12]: started with pid [21417]
process[twist_mux-13]: started with pid [21442]
process[spawn_husky_model-14]: started with pid [21443]
[WARN] [1514969623.123631, 0.000000]: Master_discovery node appear not to running. Wait for topic with type 'MasterState.
[ INFO] [1514969623.599228095]: [twist_marker_server] Initialized.
[gazebo-2] process has died [pid 21370, exit code 255, cmd /opt/ros/kinetic/lib/gazebo_ros/gzserver -e ode /home/aditya/catkin_ws/src/husky-kinetic-devel/husky_gazebo/worlds/clearpath_playpen.world __name:=gazebo __log:=/home/aditya/.ros/log/9542e33e-f063-11e7-9b71-4cbb582cd2fb/gazebo-2.log].
log file: /home/aditya/.ros/log/9542e33e-f063-11e7-9b71-4cbb582cd2fb/gazebo-2*.log
`  
Is there any solution for this error?

dniewinski, 2018-03-21:
These issues have been fixed in the kinetic-devel branch. When building from source, I recommend you use rosdep to handle installing all of your dependencies.  Kinetic is currently in development and we are are hoping to release it within the next month or two.

tonybaltovski, 2018-04-13:
We recently released the Husky packages for Kinetic. They aren't in the main repository but in [shadow-fixed](http://wiki.ros.org/ShadowRepository). That would be the best way to get the simulation running till they are synced.

closed 2018-04-13 by tonybaltovski