husky/husky #67: kinetic apt packages

mauricefallon, 2017-08-23:
Is there a plan to produce ROS Kinetic packages e.g. ros-kinetic-husky-description?
Or is it assumed people will compile from source?

- thanks!

dniewinski, 2018-03-21:
Sorry for the delay on this. Kinetic is currently in development and we are are hoping to release it within the next month or two. When that happens, pre-compiled debs will be made available as well.  This issue will be updated when that is completed.

tonybaltovski, 2018-04-13:
We recently released the Husky packages for Kinetic. They aren't in the main repository but in [shadow-fixed](http://wiki.ros.org/ShadowRepository). That would be the best way to get the package till they are synced.

closed 2018-04-13 by tonybaltovski