husky/husky #73: Dual UR5 Husky Velodyne frames not part of URDF

wxmerkt, 2018-03-06:
Hi,
Opening this ticket here since the @DualUR5Husky repository doesn't allow the creation of issues.

**Problem:** The bulkhead CAD model includes the VLP16, i.e. there is no frame information part of the URDF. Thus, no ``velodyne`` frame exists and the sensor data which is published cannot be displayed or used otherwise.

**Solution:** The velodyne frame could be added as a fixed transform from the bulkhead. Happy to do this, could you please pass the correct offsets?

Thank you very much,
Wolfgang

cc @cmower

dniewinski, 2018-03-07:
Hi.  Thanks for noticing this.  I've added the frame to the URDF in the dual_ur5_husky branch.

wxmerkt, 2018-03-07:
Thanks, tested and appears to work. Please note that you added it to the branch here which is different from the branch/integration in @DualUR5Husky (I cherry-picked and it works).

closed 2018-03-07 by wxmerkt