husky/husky #78: Husky wheels in RViz error

SeanGuoXuan, 2018-04-13:
I am following the instruction on how to[ setup Husky in Kinetic](https://answers.ros.org/question/256756/how-to-install-husky-simulator-in-kinetic/). However, when I start gazebo and RViz, Husky wheels appear display correctly in gazebo but fail in RViz. 4 wheels missing and wheel_link is giving error:
front_left_wheel_link No transform from [front_left_wheel_link] to [odom]

I don't find any problem in husky.urdf.xacro or wheel.urdf.xacro. What could cause this bug?

tonybaltovski, 2018-04-13:
We recently released the Husky packages for Kinetic.  They aren't in the main repository but in [shadow-fixed](http://wiki.ros.org/ShadowRepository).  Could you try those instead?

tonybaltovski, 2018-05-14:
Closing due to inactivity, please feel free to reopen if it is still an issue.

closed 2018-05-14 by tonybaltovski

shehancaldera, 2019-03-29:
I like to re-open issue #78 as I'm facing the same problem with simulating multiple huskies. I can't see their wheel links in the tf tree as well. Appreciate any help.