husky/husky #79: husky move base demo: navigation goal failed to move

stevealbertwong, 2018-04-15:
hi i followed this demo:
http://wiki.ros.org/husky_navigation/Tutorials/Husky%20Move%20Base%20Demo
but when i put in navigation goal husky does not move 
i am running indigo 

in rviz there is robot model error e.g. 
front_left_wheel_link
No transform from [front_left_wheel_link] to [odom]

rear_right_wheel_link
No transform from [rear_right_wheel_link] to [odom]

[image]
[image]

tonybaltovski, 2018-05-14:
Is this a real robot or a simulation?

stevealbertwong, 2018-05-26:
simulation

dniewinski, 2018-06-22:
Looking in your one window, it looks like your controller spawner couldn't create the controller properly when launch the gazebo simulation. If you launch just the simulation, does it start without errors?  If it does, can you send it a cmd_vel from another terminal?

tonybaltovski, 2018-10-17:
Closing due to inactivity.  Please feel free to re-open if issue persists.

closed 2018-10-17 by tonybaltovski

cognitiveRobot, 2019-04-25:
Hi, I am having the same issue.

> OS: Ubuntu 16.04(4.15.0-47-generic )
> ROS: kinetic. 

Can you please help to fix?
[image]

cognitiveRobot, 2019-04-25:
After updating husky packages from ros-shadow-fixed repository, it's now working. :) Go to [#78]