husky/husky #81: Does Husky_ur5 desciption package support Kinetic ROS version ???

Abduoit, 2018-04-27:
Hi

I am running husky_ur5 packages with ubuntu 14.04 and ROS `indigo`, and everything works fine. 

Now, I have to switch to use Ubuntu 16.04 with ROS `kinetic`. Is everything in husky_ur5 compatible with this combination. 

Thanks

tonybaltovski, 2018-05-14:
@Abduoit unforeuntly, kinetic is still not officially supported but will be soon!  The manipulator integrations have changed to use https://github.com/husky/husky_manipulation

Let us know if any questions.

tonybaltovski, 2018-10-17:
Kinetic has been released. 
Closing due to inactivity.  Please feel free to re-open if issue persists.

closed 2018-10-17 by tonybaltovski

Abduoit, 2020-07-18:
Hi 
Is there a `HuskyUR5 manipulation` package (`kinetic`) works on gazebo? the link below doesn't have gazebo !!?

https://github.com/husky/husky_manipulation

dniewinski, 2020-07-20:
@Abduoit No, gazebo is not supported.

Abduoit, 2020-07-20:
Thanks for replying @dniewinski 

Do you think there's a way to launch `HuskyUR5` on `Gazebo` with `ROS Kinetic`?. 
I have been trying to make it work, but I keep getting issues with UR5 controller, and I can't make it in normal position, as seen in the attached pic. 

[image]

dniewinski, 2020-07-20:
You can try, but I think you will have some issues.  The arm seems to work when it is mounted to something stationary, but has a lot of odd issues when mounted to something mobile.