husky/husky #83: amcl doesnot avoid obstacles actively

hoppss, 2018-05-14:
Hi, 

Now amcl will not avoid obstacles if there are new obstacle in the workspace that does not log in gmapping maps.

How should we modify this pkg ?

tonybaltovski, 2018-05-14:
@hoppss can you give more details about what you are using?  (ie real vs sim)  Also, are you using indigo or kinetic?  I see to explicit issues with this on the navigation repo.

hoppss, 2018-05-15:
Hi ,
sorry. 
We bought both  husky and jackal . I found jackal ( not husky, sorry) didnot avoid obstacles when i run jackal repo 's  navigation package  with a map that i made by gmapping.
It looks like the local planner doesnot work.

tonybaltovski, 2018-05-23:
@hoppss are the laser's frame and topic correct?

hoppss, 2018-06-19:
@tonybaltovski you are right!
Thank you very much. May you have a good day.

tonybaltovski, 2018-06-19:
@hoppss was this integrated by us?

hoppss, 2018-06-20:
no.
It is well to simulation with jackal gazebo model.
I clone it to my pc to run it with a real jackal, I didnot modify the frame parameters completely.

dniewinski, 2018-06-22:
@hoppss Glad to hear everything is working.  Closing the ticket

closed 2018-06-22 by dniewinski