husky/husky #85: AMCL Demo Pose Estimate?

aarontan94, 2018-05-19:
Hello,

I have a question about the initiation process of the AMCL Demo.
In the tutorial, it mentions to do: "Use the 2D Pose Estimate tool in the top toolbar to give amcl an initial pose estimate. Without an initial estimate, the Monte Carlo localization approach is unlikely to converge the correct pose. "

I am wondering, for simulation - is this as simple as using the 2D Pose Estimate arrow and select the husky's initial location in RVIZ and point the arrow in the direction of the heading angle? How does this process work for physical experimentation?

Thank you,
Aaron

aarontan94, 2018-05-23:
@tonybaltovski I am more wondering about the specifics. How do I actually use the 2D pose estimate tool? Is it as I described?

tonybaltovski, 2018-05-23:
@aarontan94 Yup, just like that. By clicking the `rviz/SetInitialPose` you are publishing to AMCL's `initialpose` topic. (for reference: http://wiki.ros.org/amcl?distro=kinetic)

aarontan94, 2018-05-23:
@tonybaltovski thank you!

closed 2018-05-23 by aarontan94