husky/husky #87: Setting back LOGITECH control

luiscarlosfeliz, 2018-05-29:
Hi, you made me lose sometime until I realized you made this commit: 
`862fd9c7df6bf3b57f9895f94e59003bcc460426
`
How do I set back the controller, I'm not being able to set the environment variable.

thank you

luiscarlosfeliz, 2018-05-30:
teleop node just like this:
```

 <!--Teleop -->
    <include file="$(find husky_control)/launch/teleop.launch" />
```

the husky nodes in general like this:

```
<arg name="port" default="$(optenv HUSKY_PORT /dev/HUSKY)" />
  <arg name="robot_namespace" default="/"/>
  <arg name="laser_enabled" default="$(optenv HUSKY_LMS1XX_ENABLED true)"/>
  <arg name="kinect_enabled" default="$(optenv HUSKY_KINECT_ENABLED false)"/>
  <arg name="urdf_extras" default="$(optenv HUSKY_URDF_EXTRAS)"/>

  <param name="robot_description" command="$(find xacro)/xacro '$(find husky_description)/urdf/husky.urdf.xacro'
    --inorder
    robot_namespace:=$(arg robot_namespace)
    laser_enabled:=$(arg laser_enabled)
    kinect_enabled:=$(arg kinect_enabled)
    urdf_extras:=$(arg urdf_extras)
    " />

	<node name="robot_state_publisher" pkg="robot_state_publisher" type="robot_state_publisher" />
    
	<!-- Laser LMS 151 -->
	<arg name="host" default="192.168.0.5" />
	<node pkg="lms1xx" name="lms1xx" type="LMS1xx_node" output="screen">
	<param name="host" value="$(arg host)" />
	</node>
        <node pkg="tf" type="static_transform_publisher" name="laser_base_link" args="0.39 0 0.36 0 0 0  /base_link /laser 10"/>
    <!-- Husky harware driver -->
    <node pkg="husky_base" type="husky_node" name="husky_node">
        <rosparam subst_value="true">
            port: $(arg port)

            control_frequency: 10.0
            diagnostic_frequency: 1.0

            max_acceleration: 0.5
            max_speed: 0.5
            wheel_diameter: 0.3555
            polling_timeout: 10.0
        </rosparam>
    </node>
```

tonybaltovski, 2018-05-30:
Where do you do `export HUSKY_LOGITECH=1`?  Is `teleop.launch` started by `robot_upstart` or in the foreground?

dniewinski, 2018-06-18:
As Tony said, you should be able to re-enable the Logitech if you are using Kinetic by exporting HUSKY_LOGITECH=1 (as stated above).  You will want to do this in /etc/ros/setup.bash.  If you've done this you can tell if it started up properly by running "rosparam get /joy_teleop/joy_node/dev".  If this says "/dev/input/js0", then it is trying to use the Logitech properly.

If that is the case, you will need to make sure the controller is connected and showing up as js0.  If you have python-ds4drv installed at the same time, it will usually take js0, which moves the Logitech controller to js1.  Try uninstalling python-ds4drv if it is installed.

dniewinski, 2018-06-20:
Please let us know if you have resolved the issue and we can close this ticket.

dniewinski, 2018-06-22:
As we have not heard back and there is an environment variable to solve this issue, we will be closing this ticket.  If you are still having issues after following the steps above, please open another ticket.

closed 2018-06-22 by dniewinski