husky/husky #88: husky_navigation has move_base publishing to wrong vel topic

bfalacerda, 2018-06-11:
I'm trying to run the basic move_base stuff using the ``husky_navigation`` launch files but ``move_base`` is publishing to ``/cmd_vel``. This doesn't make the robot move due to the mux setup.  It's happening since the commit below:

https://github.com/husky/husky/commit/5a5f0de2b4e00e4acb46928004fcce10404a92b6

The remap removed above is required, or am I missing something?

tonybaltovski, 2018-06-12:
That is certainly odd, that remap was for internal testing.  The mux should respond to `cmd_vel` according to the config [here](https://github.com/husky/husky/blob/kinetic-devel/husky_control/config/twist_mux.yaml#L11).  I will test this as soon as I can.  Is there any chance something else is publishing to the `twist_mux` or that there is a clock skew between the robot's computer and what is sending the `cmd_vel`?  Thanks for reporting this!

bfalacerda, 2018-06-13:
ah! sorry, my bad, we misconfigured that file... all is good. Thanks for the quick reply.

closed 2018-06-13 by bfalacerda