husky/husky #99: Strange model behaviour in gazebo with certain URDF configurations

ManolisCh, 2018-10-19:
Hi, 

I was giving a go to gazebo simulated husky by running `roslaunch husky_gazebo husky_playpen.launch` and  `roslaunch husky_control teleop.launch`. I have noticed a strange behaviour when I command the robot to turn in place but also in movement.

Basically the robot would not turn in place most of the times and it would slide slow or faster towards one random direction. This is with linear velocity 0 and any amount of angular velocity (with twist message). Note that this appears only when I either disable the laser or I  set  `kinect_enabled` to true. In the case I only use the model with the `laser_enabled` to true the husky is able to turn in place and his navigation behaviour is normal/as expected. 

1. When I run the `husky_playpen.launch` with only the laser enabled (i.e. when husky behaviour is as expected) I get the following warning which might have something to do with it: 

`Warning [parser_urdf.cc:1232] multiple inconsistent <gravity> exists due to fixed joint reduction overwriting previous value [true] with [false].`

2. When I run gazebo with both laser and kinect I get the warning twice and the robot does not move as expected. It seems to me some correct settings are over written:

```
Warning [parser_urdf.cc:1232] multiple inconsistent <gravity> exists due to fixed joint reduction overwriting previous value [true] with [false].
Warning [parser_urdf.cc:1232] multiple inconsistent <gravity> exists due to fixed joint reduction overwriting previous value [false] with [true].
```

3. If I run gazebo without kinect and laser the strange behaviour is still there but I do not get any of the above messages.

I also get regardless of the laser enabled or disabled the following:

`[ WARN] [1539887432.825685562, 1298.690000000]: updateConfig() called on a dynamic_reconfigure::Server that provides its own mutex. This can lead to deadlocks if updateConfig() is called during an update. Providing a mutex to the constructor is highly recommended in this case. Please forward this message to the node author.`

Please advice, as this behaviour really messes with teleoperation or autonomous navigation.

thanks for you time,
Manolis

dkati, 2019-04-25:
i have that too ->

[ WARN] [1539887432.825685562, 1298.690000000]: updateConfig() called on a dynamic_reconfigure::Server that provides its own mutex. This can lead to deadlocks if updateConfig() is called during an update. Providing a mutex to the constructor is highly recommended in this case. Please forward this message to the node author.

civerachb-cpr, 2020-06-02:
1) This is because the LMS 1xx package specifies to turn gravity off in its Gazebo tag.  You can modify that package's sick_lms1xx.urdf.xacro file to change <turnOffGravity> from true to false. This will suppress the warning, but seems to have the side-effect of further breaking the simulation.  I would advise ignoring this warning; it's just notifying you that gravity is acting on some links but not others.

2) Same as above, but with the Husky's own kinect_camera.urdf.xacro file.  Again, I wouldn't advise modifying this file to suppress the warning.

3) I've been unable to replicate this on my machine.  If I run the simulation and then run
`rostopic pub /husky_velocity_controller/cmd_vel geometry_msgs/Twist "linear:
  x: 0.0
  y: 0.0
  z: 0.0
angular:
  x: 0.0
  y: 0.0
  z: 0.3" -r 10`
the robot moves continuously.  It does drift away from the origin a bit (see https://github.com/husky/husky/issues/105), but I'm not seeing the robot start and stop like you describe.
This particular warning also does not appear to occur in ROS Melodic.  Because Kinectic is going EOL in the near future you may want to try upgrading to Melodic and see if that resolves the issue.

civerachb-cpr, 2020-09-09:
Closing due to inactivity.

closed 2020-09-09 by civerachb-cpr