husky/husky #100: Melodic roslaunch failure

abhileshborode, 2018-10-20:
On running 

roslaunch husky_gazebo husky_empty_world.launch
I get the following error
... logging to /home/abhilesh/.ros/log/7779b0f2-d4c0-11e8-9d77-bc8556ce29bb/roslaunch-abhilesh-Ubuntu-30317.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
resource not found: lms1xx
ROS path [0]=/opt/ros/melodic/share/ros
ROS path [1]=/home/abhilesh/husky/src
ROS path [2]=/opt/ros/melodic/share
when processing file: /home/abhilesh/husky/src/husky_description/urdf/accessories/sick_lms1xx_mount.urdf.xacro
included from: /home/abhilesh/husky/src/husky_description/urdf/husky.urdf.xacro
RLException: while processing /home/abhilesh/husky/src/husky_gazebo/launch/spawn_husky.launch:
while processing /home/abhilesh/husky/src/husky_control/launch/control.launch:
while processing /home/abhilesh/husky/src/husky_description/launch/description.launch:
Invalid <param> tag: Cannot load command parameter [robot_description]: command [/opt/ros/melodic/lib/xacro/xacro '/home/abhilesh/husky/src/husky_description/urdf/husky.urdf.xacro'     --inorder     robot_namespace:=/     laser_enabled:=true     kinect_enabled:=false     urdf_extras:=     ] returned with code [2]. 

Param xml is <param command="$(find xacro)/xacro '$(find husky_description)/urdf/husky.urdf.xacro'     --inorder     robot_namespace:=$(arg robot_namespace)     laser_enabled:=$(arg laser_enabled)     kinect_enabled:=$(arg kinect_enabled)     urdf_extras:=$(arg urdf_extras)     " name="robot_description"/>
The traceback for the exception was written to the log file

Any leads on a way around these errors ?

tonybaltovski, 2018-10-21:
You are missing the `lms1xx` package.

tonybaltovski, 2018-10-23:
Looks like you a building a different branch or older commits, none of the ur stuff should be in the latest kinetic-devel.  As for the rest, I don't see them release for melodic.  Maybe, poke their maintainers?

No worries!

closed 2018-11-21 by tonybaltovski