husky/husky #102: gazebo_ros_control pid_gains not set Error

FahadRazaSJTU, 2018-11-30:
[image]

Hello guys, 
I am trying to use husky_gazebo in ROS-Lunar. However, when I run the husky_empty_world.launch I get the error which is shown in the attached screenshot for all of the four wheels. After searching through internet I have found the solution, which is to add pid_gains manually in control.yaml file.

I am still wondering why it happened? 

Thank you for your help.

skyofyao, 2019-02-25:
I had the same question. That was because when initializing RobotHWSim, the robot_ros_control tries to load the pid_gains. The code and corresponding comments are [here](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/e42cc26a473733b05abc648181a6de1286105ed6/gazebo_ros_control/src/default_robot_hw_sim.cpp#L218). Basically, if pid parameters were found, they will be used to control the joints (velocity or position) by effort.  Otherwise, the joints will be controlled with gazebo methods.

FahadRazaSJTU, 2019-02-26:
Hi @skyofyao  you are right. Thank you for sharing the code.

civerachb-cpr, 2020-06-02:
Please see the comment I left here for a more detailed explanation of why this error is not actually an error: https://github.com/husky/husky/issues/110

closed 2020-06-02 by civerachb-cpr