husky/husky #105: linear motion commanding only rotation

jocacace, 2019-02-06:
Hi guys,

I am addressing a problem controlling the husky robot in the gazebo simulator.
I just spawn a husky robot in an empty gazebo world. I am trying to rotate on the place ( I suppose that I am able to do with this platform since it has a differential traction ). For this reason, I just published a geometry_msgs::Twist message filling the .angular.z field with a fixed velocity (let's say 0.3 rad/s).
Sadly, what's happen is that the robot start to glide in the lateral directions. After 3 complete rotation on the place the robot is far of 1, 1.5 meters from the starting position... 

Is this a common behaviour of this platform for this kind of motion??

Thank you, 
Cheers.

epiception, 2019-06-06:
I'm having the same problem. Any quick solutions for this? Any changes that could be made in the controller yaml files?

paulsebastian97, 2019-10-07:
Even I phased this problem. Could anyone help?

civerachb-cpr, 2020-06-02:
This looks like it may be a glitch in Gazebo's physics.  Because Husky's wheels are fixed, when the robot turns on the spot they will skip and skid across the ground, which can result in some lateral movement.

civerachb-cpr, 2020-06-03:
As noted in #99, you may also be able to work-around this issue by moving the Husky's centre of inertia backwards slightly.  This seems to counteract the imbalance that the lidar introduces.

And1210, 2020-06-04:
I was able to change the friction on the wheels which seems to make the robot turn more smoothly. A description of what was done can be found here: https://github.com/husky/husky/issues/120#issuecomment-638921486

civerachb-cpr, 2020-06-04:
Closing this issue, as it seems to be a duplicate of #120 (or at least the discussion is largely a duplication).