husky/husky #112: Error launching husky_base on Raspberry Pi

luissantos27, 2019-08-08:
I get this error when I launch the husky_base on a raspberry PI with ROS kinetic. 

EXCEPTION: TransportException 3: Unacknowledged send
[ERROR] [1565258419.702807681]: Error configuring velocity and accel limits: Unacknowledged send

The USB port has permissions and I changed it on the launch file. I compiled the husky software by source with the branch "kinetic-devel" 

I saw that there was a bug with arm platforms and I tried to correct it with no success: 

https://answers.ros.org/question/215206/husky-bringup-on-an-arm-platform/

Here is my terminal log:

`pi@raspberrypi:~/catkin_ws $ roslaunch husky_base base.launch 
... logging to /home/pi/.ros/log/e168a6e8-b93b-11e9-8921-b827eb2ea439/roslaunch-raspberrypi-3844.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

inconsistent namespace redefinitions for xmlns:xacro:
 old: http://ros.org/wiki/xacro
 new: http://www.ros.org/wiki/xacro (/home/pi/catkin_ws/install_isolated/share/lms1xx/urdf/sick_lms1xx.urdf.xacro)
started roslaunch server http://172.16.47.94:41841/

SUMMARY
========

PARAMETERS
 * /diagnostic_aggregator/analyzers/husky/expected: ['husky_base: pow...
 * /diagnostic_aggregator/analyzers/husky/find_and_remove_prefix: husky_base: 
 * /diagnostic_aggregator/analyzers/husky/path: Husky A200
 * /diagnostic_aggregator/analyzers/husky/timeout: 5.0
 * /diagnostic_aggregator/analyzers/husky/type: diagnostic_aggreg...
 * /ekf_localization/base_link_frame: base_link
 * /ekf_localization/frequency: 50
 * /ekf_localization/imu0: imu/data
 * /ekf_localization/imu0_config: [False, False, Fa...
 * /ekf_localization/imu0_differential: True
 * /ekf_localization/imu0_queue_size: 10
 * /ekf_localization/imu0_remove_gravitational_acceleration: True
 * /ekf_localization/odom0: husky_velocity_co...
 * /ekf_localization/odom0_config: [False, False, Fa...
 * /ekf_localization/odom0_differential: False
 * /ekf_localization/odom0_queue_size: 10
 * /ekf_localization/odom_frame: odom
 * /ekf_localization/two_d_mode: True
 * /ekf_localization/world_frame: odom
 * /husky_joint_publisher/publish_rate: 50
 * /husky_joint_publisher/type: joint_state_contr...
 * /husky_node/control_frequency: 10.0
 * /husky_node/diagnostic_frequency: 1.0
 * /husky_node/max_acceleration: 3.0
 * /husky_node/max_speed: 1.0
 * /husky_node/polling_timeout: 10.0
 * /husky_node/port: /dev/ttyUSB0
 * /husky_node/wheel_diameter: 0.3302
 * /husky_velocity_controller/angular/z/has_acceleration_limits: True
 * /husky_velocity_controller/angular/z/has_velocity_limits: True
 * /husky_velocity_controller/angular/z/max_acceleration: 6.0
 * /husky_velocity_controller/angular/z/max_velocity: 2.0
 * /husky_velocity_controller/base_frame_id: base_link
 * /husky_velocity_controller/cmd_vel_timeout: 0.25
 * /husky_velocity_controller/enable_odom_tf: False
 * /husky_velocity_controller/estimate_velocity_from_position: False
 * /husky_velocity_controller/left_wheel: ['front_left_whee...
 * /husky_velocity_controller/linear/x/has_acceleration_limits: True
 * /husky_velocity_controller/linear/x/has_velocity_limits: True
 * /husky_velocity_controller/linear/x/max_acceleration: 3.0
 * /husky_velocity_controller/linear/x/max_velocity: 1.0
 * /husky_velocity_controller/pose_covariance_diagonal: [0.001, 0.001, 0....
 * /husky_velocity_controller/publish_rate: 50
 * /husky_velocity_controller/right_wheel: ['front_right_whe...
 * /husky_velocity_controller/twist_covariance_diagonal: [0.001, 0.001, 0....
 * /husky_velocity_controller/type: diff_drive_contro...
 * /husky_velocity_controller/velocity_rolling_window_size: 2
 * /husky_velocity_controller/wheel_radius_multiplier: 1.0
 * /husky_velocity_controller/wheel_separation_multiplier: 1.875
 * /joy_teleop/joy_node/autorepeat_rate: 20
 * /joy_teleop/joy_node/deadzone: 0.1
 * /joy_teleop/joy_node/dev: /dev/input/ds4x
 * /joy_teleop/teleop_twist_joy/axis_angular: 0
 * /joy_teleop/teleop_twist_joy/axis_linear: 1
 * /joy_teleop/teleop_twist_joy/enable_button: 4
 * /joy_teleop/teleop_twist_joy/enable_turbo_button: 5
 * /joy_teleop/teleop_twist_joy/scale_angular: 1.4
 * /joy_teleop/teleop_twist_joy/scale_linear: 0.4
 * /joy_teleop/teleop_twist_joy/scale_linear_turbo: 2.0
 * /robot_description: <?xml version="1....
 * /rosdistro: kinetic
 * /rosversion: 1.12.13
 * /twist_mux/locks: [{'topic': 'e_sto...
 * /twist_mux/topics: [{'topic': 'joy_t...

NODES
  /joy_teleop/
    joy_node (joy/joy_node)
    teleop_twist_joy (teleop_twist_joy/teleop_node)
  /
    base_controller_spawner (controller_manager/spawner)
    diagnostic_aggregator (diagnostic_aggregator/aggregator_node)
    ekf_localization (robot_localization/ekf_localization_node)
    husky_node (husky_base/husky_node)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    twist_marker_server (interactive_marker_twist_server/marker_server)
    twist_mux (twist_mux/twist_mux)

ROS_MASTER_URI=http://172.16.47.94:11311

process[husky_node-1]: started with pid [3866]
process[base_controller_spawner-2]: started with pid [3867]
process[ekf_localization-3]: started with pid [3868]
process[twist_marker_server-4]: started with pid [3869]
process[robot_state_publisher-5]: started with pid [3881]
process[twist_mux-6]: started with pid [3886]
process[joy_teleop/joy_node-7]: started with pid [3899]
process[joy_teleop/teleop_twist_joy-8]: started with pid [3912]
process[diagnostic_aggregator-9]: started with pid [3921]
[ERROR] [1565258419.148172971]: Couldn't open joystick /dev/input/ds4x. Will retry every second.
[ INFO] [1565258419.167212299]: [twist_marker_server] Initialized.
EXCEPTION: TransportException 3: Unacknowledged send
[ERROR] [1565258419.702807681]: Error configuring velocity and accel limits: Unacknowledged send
EXCEPTION: TransportException 3: Unacknowledged send
[ERROR] [1565258420.564875992]: Error configuring velocity and accel limits: Unacknowledged send
^CEXCEPTION: TransportException 3: Unacknowledged send
[ERROR] [1565258421.433556182]: Error configuring velocity and accel limits: Unacknowledged send
[diagnostic_aggregator-9] killing on exit
[joy_teleop/teleop_twist_joy-8] killing on exit
[joy_teleop/joy_node-7] killing on exit
[ekf_localization-3] killing on exit
[twist_mux-6] killing on exit
[robot_state_publisher-5] killing on exit
[husky_node-1] killing on exit
[twist_marker_server-4] killing on exit
[base_controller_spawner-2] killing on exit
[WARN] [1565258422.156846]: Controller Spawner couldn't find the expected controller_manager ROS interface.
EXCEPTION: TransportException 3: Unacknowledged send
EXCEPTION: TransportException 3: Unacknowledged send
EXCEPTION: TransportException 3: Unacknowledged send
EXCEPTION: TransportException 3: Unacknowledged send
EXCEPTION: TransportException 3: Unacknowledged send
EXCEPTION: TransportException 3: Unacknowledged send
EXCEPTION: TransportException 3: Unacknowledged send
EXCEPTION: TransportException 3: Unacknowledged send
EXCEPTION: TransportException 3: Unacknowledged send
`

tonybaltovski, 2019-08-21:
Did you install the udev rule for the USB-to-Serial adapter?

luissantos27, 2019-08-26:
> Did you install the udev rule for the USB-to-Serial adapter?

I don't think so. How could I do this? I am not getting an error with the serial connection, this only happens when I try to set the maximum velocity values.

luissantos27, 2019-08-27:
> You can install it using `sudo cp $(rospack find husky_bringup)/udev/* /etc/udev/rules.d/ sudo 
udevadm control --reload-rules && sudo service udev restart && sudo udevadm trigger`.

This did not work...

> Are you able to drive the Husky when you send non-maximum velocity values?

No, I'm not. The husky_node dies: 

`process[husky_node-1]: started with pid [20703]
process[base_controller_spawner-2]: started with pid [20704]
process[ekf_localization-3]: started with pid [20705]
process[twist_marker_server-4]: started with pid [20706]
process[robot_state_publisher-5]: started with pid [20707]
process[twist_mux-6]: started with pid [20708]
process[joy_teleop/joy_node-7]: started with pid [20709]
process[joy_teleop/teleop_twist_joy-8]: started with pid [20710]
process[diagnostic_aggregator-9]: started with pid [20711]
[ INFO] [1566901653.626636996]: Connecting to Husky on port /dev/ttyUSB0...
[ INFO] [1566901653.629900368]: Connected
[ERROR] [1566901653.634078840]: Couldn't open joystick /dev/input/ds4x. Will retry every second.
[ INFO] [1566901653.730677247]: [twist_marker_server] Initialized.
EXCEPTION: TransportException 3: Unacknowledged send
terminate called after throwing an instance of 'clearpath::TransportException*'
[husky_node-1] process has died [pid 20703, exit code -6, cmd /home/pi/catkin_ws/install_isolated/lib/husky_base/husky_node __name:=husky_node __log:=/home/pi/.ros/log/8d84b7fc-c8ad-11e9-8135-b827eb2ea439/husky_node-1.log].
log file: /home/pi/.ros/log/8d84b7fc-c8ad-11e9-8135-b827eb2ea439/husky_node-1*.log
`

luissantos27, 2019-10-03:
I have tries to do the same using a jeston nano device, with all the packages were installed via apt using ubuntu official ppas and I'm getting exactly the same error! The robot works with a normal computer..

Alex-Beh, 2022-03-31:
@tonybaltovski I am facing the same issue and the udev rules don't help... Any other hints?

civerachb-cpr, 2022-03-31:
@Alex-Beh are you still using Kinetic? Or are you using a newer ROS distro like Melodic or Noetic?

We changed the location of some of the udev rules in Melodic to reduce duplication across multiple robot models.

Alex-Beh, 2022-03-31:
@civerachb-cpr I am using noetic branch. And I realize if I launch seperately then the issue won't happen.... (yeah it don't make sense.... will continue do testing and update here again tomorrow)

civerachb-cpr, 2022-03-31:
If you're able to launch separately but the Systemd job isn't working the likeliest explanation is a permissions problem with something on the system.

What happens if you run
```
sudo systemctl stop ros
sudo /usr/sbin/ros-start
```

tonybaltovski, 2022-03-31:
Could you also run ` ls -lisah /dev/*` and provide the output.

closed 2022-04-29 by tonybaltovski