husky/husky #113: Wrong IMU topic for EKF with UM6/7 IMUs?

peci1, 2019-08-20:
Line https://github.com/husky/husky/blob/862fd9c7df6bf3b57f9895f94e59003bcc460426/husky_control/config/localization.yaml#L18 refers to topic `imu/data` which doesn't exist at least on our Husky A2 with UM6 IMU. We have topic `imu/data_raw`, however. 

I see the `imu/data` topic in configs for both UM6 and UM7, whereas the microstrain nodes seem to publish on `imu/data`. Do we do something wrong, or is just everybody using Husky with UM* IMU running EKF without the IMU?

Also, setting the [`two_d_mode`](https://github.com/husky/husky/blob/862fd9c7df6bf3b57f9895f94e59003bcc460426/husky_control/config/localization.yaml#L5) by default seems like something really unexpected to me...

tonybaltovski, 2019-08-21:
The data gets filtered using `imu_filter_madgwick` which subscribes in `imu/data_raw` and publishes `imu/data`: https://github.com/husky/husky/blob/kinetic-devel/husky_bringup/launch/um6_config/um6.launch#L23

peci1, 2019-08-23:
Thanks, Tony. Our Husky went now for a trip home which will take some time, so I'll get back to you later when we receive it again. Or @rouceto1 might chime in.

When testing, I'm pretty sure that in rqt_graph, I saw the IMU filter publish to some topic which was not `imu/data`, and we wondered why nobody is subscribing the filtered IMU... So let's wait and I'll check then if we don't have some local modifications which would break the publishing chain...

peci1, 2019-09-10:
Ok, I think I got it. 

It seems the um6 driver publishes `mag` as the old `Vector3Stamped` type. This was handled by the `use_magnetic_field_msg: false` config in madgwick filter config. However, the filter stopped supporting this option some time ago: https://github.com/ccny-ros-pkg/imu_tools/blob/melodic/imu_filter_madgwick/CHANGELOG.rst#115-2017-05-24 . This means the filter subscribes the right topic but with wrong type, which is confirmed by my console:

    [ERROR] [1568128900.059345260]: Client [/imu_manager2] wants topic /imu/mag to have datatype/md5sum [sensor_msgs/MagneticField/2f3b0b43eed0c9501de0fa3ff89a45aa], but our version has [geometry_msgs/Vector3Stamped/7b324c7325e683bf02a9b14b01090ec7]. Dropping connection.

And the madgwick filter hangs on waiting for a correct mag message.

Now the question is, what is the correct fix for this? Is there a node that would translate the old mag type to the new one?

tonybaltovski, 2019-09-10:
@peci1  Are you using a um6 or um7?  Which ROS distro are you targeting?

rouceto1, 2019-09-11:
@tonybaltovski we are using um6 with melodic distro.

tonybaltovski, 2019-09-11:
This should fix your issue: https://github.com/ros-drivers/um6/tree/mag_msg
I will work on getting the merged soon.  Let me know if it works for you.

tonybaltovski, 2019-09-17:
It would and I will for Melodic since I released it already.  However, like `imu_filter_madgwick` it should eventually follow the REP.

tonybaltovski, 2019-09-30:
@peci1 I have updated to um6 and um7.  They have also been bloomed.  I will be bloom the Husky asap.  This should solve the issue but I will wait for you to confirm before I close it.

closed 2020-03-11 by tonybaltovski