husky/husky #118: EXCEPTION: TransportException 3: Unacknowledged send

je5349, 2019-11-21:
We use to logitech's extreme 3d pro for control husky.
We change from "$(optenv HUSKY_LOGITECH 0)" to "$(optenv HUSKY_LOGITECH 1)" in husky_control/launch/teleop.launch file. (line 9 and line 13)
And we change "teleop.yaml" file in "husky_control/config" as shown below.

----------------------------------
teleop_twist_joy:
  axis_linear: 1
  scale_linear: 0.4
  scale_linear_turbo: 1.0

  axis_angular: 2
  scale_angular: 0.6
  scale_angular_turbo: 1.0

  enable_button: 0

  enable_turbo_button: 1

joy_node:
  deadzone: 0.1
  autorepeat_rate: 20
------------------------------------

And then, we run "roslaunch husky_base base.launch" and no error.
But, when we operate joystick, it occurs error as shown below.

-------------------------------------
Error in serial write
Error in serial write
Error in serial write
Error in serial write
EXCEPTION: TransportException 3: Unacknowledged send
terminate called after throwing an instance of 'clearpath::TransportException*'
[husky_node-2] process has died [pid 3539, exit code -6, cmd /home/aeco_lp/catkin_ws/devel/lib/husky_base/husky_node __name:=husky_node __log:=/home/aeco_lp/.ros/log/bc96267e-0c6a-11ea-a4aa-00e04c122422/husky_node-2.log].
log file: /home/aeco_lp/.ros/log/bc96267e-0c6a-11ea-a4aa-00e04c122422/husky_node-2*.log
---------------------------------

We find "Error in serial write" in the "husky_base/src/horizon_legacy/linux_serial.cpp" (line 127).
Could you tell me, how to fix it?

je5349, 2019-11-25:
We solved this problem. 
The cause was a hardware problem. 
Thank you.

closed 2019-11-25 by je5349

eseJiHeaLim, 2020-02-07:
Can you tell me what hardware problem?
My problem is same with you.

je5349, 2020-02-18:
> Can you tell me what hardware problem?
> My problem is same with you.

I plugged the joystick into a USB hub that was connected to the upper control embedded board.
Since then I have noticed that the USB hub is running out of power.
I gave the USB hub an external power source and it worked fine afterward.