husky/husky #123: Husky Laser and PC2  Topic Conflict

rmerriam, 2020-02-23:
Ubuntu 18.04
ROS Melodic
Husky Tutorial in simulation with Gazebo

This caused me some confusion so thought I'd mention it here. 

Run the command:
```
roslaunch scout_gazebo scout_playpen.launch kinect_enabled:=true laser_enabled:=true
```
The `/scan` topic reports both the actual laser and the kinect point cloud simulated laser on the topic but with different frames, i.e. base_link and laser_base, and values, i.e. base_link is at lower point on objects. 

Might just need explanation in the tutorial rather than considering it a bug needing a fix in the launch files. The latter would be either disallowing both laser and kinect to be enabled or changing the topic for the kinect simulated laser.  In `spawn_husky.launch` I changed the kinect laser topic.

civerachb-cpr, 2020-06-02:
Good catch.  With the Kinect being discontinued for some time this is less of an urgency, but we'll look into adding this to our documentation.

civerachb-cpr, 2020-06-02:
Didn't mean to close this one before. Clicked the wrong button. Sorry!

civerachb-cpr, 2020-06-03:
Most of the necessary changes should be reflected here: https://github.com/husky/husky/pull/131

The Kinect's "laserscan" should now be on the /kinect/scan topic, instead of /scan.  If you're using the Kinect for the gmapping demo you can specify the scan topic to listen to when launching the demo; it defaults to /scan, but can be overridden.

Because the Kinect has been discontinued I've also added support for the Intel RealSense D400 series as a replacement depth camera.  Mostly the same/similar setup as the kinect (realsense_enabled:=true, laserscan data publishes to /realsense/scan).  There's also support for a physical RealSense in addition to the simulated one added to the husky_bringup package.

civerachb-cpr, 2020-06-04:
Should be fixed now.

closed 2020-06-04 by civerachb-cpr