husky/husky #124: Driving husky in gazebo environment

ZakariaChekakta, 2020-02-25:
Hi, 
I am testing the husky robot using Gazebo 9 and ROS melodic when I move the robot forward or backward, it working properly, but turning the robot to the left or the right, I noticed a strange behavior in the robot movement, Is that behavior normal or there is a bug needs to be fixed?

[image]

h3ct0r, 2020-03-21:
I'm having the same problem (see video [https://youtu.be/yKhHC9QsZZY](ur)) This is also described in other issues such as [https://github.com/husky/husky/issues/120](url) and [https://github.com/husky/husky/issues/105](url). But so far the Clearpath team has not responded to these issues.

@tonybaltovski Hi! Can you take a look at this, please? It seems that there are multiple people reporting this.

closed 2020-05-12 by ZakariaChekakta

nmarticorena, 2020-11-16:
Hi, someone manage to solve this issue? 

I already tried the comments in the other issues but the problem persists.

Any help or comment would be really appreciated.