husky/husky #128: [Question] Understanding the local planner config

adwiii, 2020-04-20:
I have been working on simulating the Husky in various simple navigation tasks. I am using the [husky_cartographer_navigation](https://github.com/husky/husky_cartographer_navigation) setup on top of the standard husky_navigation. In these navigation tasks, the robot will sometimes drive very near to obstacles and collide with them when there is plenty of open space. I am going to make a separate post to ask a more general question about that behavior, but here I want to focus on the planner config.

In tuning the navigation parameters, I have a few questions about the default values given as I try to better understand the navigation system.

In `husky_navigation/config/planner.yaml`, it is set to use DWA (line 45: `dwa:true`), but then starting on [line 75](https://github.com/husky/husky/blob/2b90c2161ead01c4f2246dd4f289b71e75b64cae/husky_navigation/config/planner.yaml#L75) multiple DWA config parameters are commented out. Cross referencing these with the [dwa_local_planner wiki](http://wiki.ros.org/dwa_local_planner), it appears most of them would be set to the default value, except for: 
```
sim_time: 2.0 # (default 1.7)
sim_granularity: 0.02 # (default 0.025)
vx_samples: 6 # (default 3)
vy_samples: 0 # (default 10)
oscillation_reset_dist: 0.25 # (default 0.05)
```
Given that the husky is not holonomic, these changes seem sensible, but they are not being used. Is there any history behind this? I believe it traces to [this commit](https://github.com/husky/husky_navigation/commit/0319dff7c713e4562d467e2d184e69b0ba6ca181), but I do not understand how these values were chosen or why they would not be used.

 In some testing I uncommented these lines and tuned the `path_distance_bias`, `goal_distance_bias`, and `occdist_scale` values to be more in line with the Trajectory Planner values for `pdist_scale`, `gdist_scale`, and `occdist_scale` respectively. I believe this is reasonable because `meter_scoring` is turned on, so the scale for both sets of parameters should be the same(?). These changes appeared to improve the stability of the navigation.

Also, not commented out, the DWA parameter for `max_vel_x` is set to 0.5 instead of 1.0 as it is in the Trajectory Planner. I would initially expect that these values should be the same. Is this intentional? Do these values have different effects in the different contexts?

civerachb-cpr, 2020-06-08:
I don't know the whole context behind that commit -- it was before my time here.  I asked around and nobody else seems to have a good explanation either, honestly.  If changing that section improves performance please feel free to submit a PR.

civerachb-cpr, 2020-09-09:
Closing due to inactivity.

closed 2020-09-09 by civerachb-cpr