husky/husky #139: DeserializationError cannot deserialize: unknown error handler name 'rosmsg'

OhmySallyGarden, 2020-08-26:
Hi guys. I am following the instruction on how to setup Husky in Kinetic in simulation. When I was running "Husky Move Base Demo",  there were several errors.
1. 
[ INFO] [1598430220.265429328, 1298.680000000]: Loaded gazebo_ros_control.
[ INFO] [1598430220.271723048, 1298.680000000]: imu plugin missing <xyzOffset>, defaults to 0s
[ INFO] [1598430220.271761029, 1298.680000000]: imu plugin missing <rpyOffset>, defaults to 0s
[spawn_husky_model-9] process has finished cleanly
log file: /home/hitero/.ros/log/6f564b46-e775-11ea-8bc9-80e82cce0294/spawn_husky_model-9*.log
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/controller_manager/spawner", line 207, in <module>
    if __name__ == '__main__': main()
  File "/opt/ros/kinetic/lib/controller_manager/spawner", line 185, in main
    resp = load_controller(name)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 435, in __call__
    return self.call(*args, **kwds)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 525, in call
    raise ServiceException("transport error completing service call: %s"%(str(e)))
rospy.service.ServiceException: transport error completing service call: receive_once[/controller_manager/load_controller]: DeserializationError cannot deserialize: unknown error handler name 'rosmsg'
[WARN] [1598430220.472726, 1298.690000]: Controller Spawner error while taking down controllers: transport error completing service call: receive_once[/controller_manager/switch_controller]: DeserializationError cannot deserialize: unknown error handler name 'rosmsg'
[image]

2. 
4 wheels missing and wheel_link is giving error in Rviz:
front_left_wheel_link No transform from [front_left_wheel_link] to [odom]
[image]

What could cause these bugs? 

thanks for you time!

civerachb-cpr, 2020-09-02:
It appears there was a bug in genpy that could cause this error.  The issue has since been resolved in genpy 0.6.13.

Can you please check if there are updates available on your system and if so, install them?

civerachb-cpr, 2020-09-03:
It appears the issue is related to https://github.com/ros/genpy/issues/132.

civerachb-cpr, 2020-09-03:
Closing as duplicate of https://github.com/husky/husky/issues/142

closed 2020-09-03 by civerachb-cpr