husky/husky #142: DeserializationError cannot deserialize: unknown error handler name 'rosmsg' --- husky_velocity_controller spawn

JackBorer, 2020-09-02:
Working on a husky and getting the following error when trying to launch the startup file which contains the velocity controller command. I really don't even know where to start with this, but I see https://github.com/husky/husky/issues/139 recently got a similar type of error message, maybe they are unrelated, I can't really say

This error began to occur exactly after I updated and upgraded using apt-get. I am running on a real Husky with kinetic and 16.04. 

The entire script from the terminal when I try to launch the start script is pasted below the launch file.

`<?xml version="1.0"?>
<launch>

  <arg name="multimaster" default="$(optenv ROBOT_MULTIMASTER false)"/>
  <arg name="simulation" default="$(optenv ROBOT_SIMULATION false)"/>
  <arg name="robot_namespace" default="$(optenv ROBOT_NAMESPACE robot)"/>

  <arg name="config_extras"
       default="$(eval optenv('HUSKY_CONFIG_EXTRAS', find('husky_control') + '/config/empty.yaml'))"/>

  <arg name="laser_enabled" default="$(optenv HUSKY_LMS1XX_ENABLED false)"/>
  <arg name="kinect_enabled" default="$(optenv HUSKY_KINECT_ENABLED false)"/>
  <arg name="urdf_extras" default="$(optenv HUSKY_URDF_EXTRAS)"/>

  <include file="$(find multimaster_launch)/launch/multimaster_robot.launch" if="$(arg multimaster)">
    <arg name="gazebo_interface" value="$(find husky_control)/config/gazebo_interface.yaml" />
    <arg name="public_interface" value="$(find husky_control)/config/public_interface.yaml" />
    <arg name="robot_namespace" value="$(arg robot_namespace)"/>
    <arg name="simulation" value="$(arg simulation)"/>
  </include>

  
  <include file="$(find husky_description)/launch/description.launch" >
    <arg name="laser_enabled" default="$(arg laser_enabled)"/>
    <arg name="kinect_enabled" default="$(arg kinect_enabled)"/>
    <arg name="urdf_extras" default="$(arg urdf_extras)"/>
  </include>

  
  <rosparam command="load" file="$(find husky_control)/config/control.yaml" />

  
  <node name="base_controller_spawner" pkg="controller_manager" type="spawner"
        args="husky_velocity_controller husky_joint_publisher" output="screen"/>

  
  

  <node pkg="interactive_marker_twist_server" type="marker_server" name="twist_marker_server" output="screen"/>

  <node name="robot_state_publisher" pkg="robot_state_publisher" type="robot_state_publisher" />

  <node pkg="twist_mux" type="twist_mux" name="twist_mux">
    <rosparam command="load" file="$(find husky_control)/config/twist_mux.yaml" />
    <remap from="cmd_vel_out" to="husky_velocity_controller/cmd_vel"/>
  </node>

  
  <rosparam command="load" file="$(arg config_extras)" />

</launch>`

` roslaunch cus_main ros_d_base.launch 
... logging to /home/administrator/.ros/log/0b5425c6-ec9a-11ea-83be-00e04c71ce5c/roslaunch-cpr-aus05-5349.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

inconsistent namespace redefinitions for xmlns:xacro:
 old: http://ros.org/wiki/xacro
 new: http://www.ros.org/wiki/xacro (/opt/ros/kinetic/share/lms1xx/urdf/sick_lms1xx.urdf.xacro)
started roslaunch server http://cpr-aus05:35725/

SUMMARY
[... 102 lines cut ...]
DynamicParams:
	Odometry parameters:
		left wheel radius: 1
		right wheel radius: 1
		wheel separation: 1.875
	Publication parameters:
		Publish executed velocity command: 0
		Publication rate: 50
		Publish frame odom on tf: 0
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/controller_manager/spawner", line 207, in <module>
    if __name__ == '__main__': main()
  File "/opt/ros/kinetic/lib/controller_manager/spawner", line 185, in main
    resp = load_controller(name)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 435, in __call__
    return self.call(*args, **kwds)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 525, in call
    raise ServiceException("transport error completing service call: %s"%(str(e)))
rospy.service.ServiceException: transport error completing service call: receive_once[/controller_manager/load_controller]: DeserializationError cannot deserialize: unknown error handler name 'rosmsg'
[INFO] [/base_controller_spawner] [shutdown]: Shutting down spawner. Stopping and unloading controllers...
[INFO] [/base_controller_spawner] [shutdown]: Stopping all controllers...
[WARN] [/base_controller_spawner] [shutdown]: Controller Spawner error while taking down controllers: transport error completing service call: receive_once[/controller_manager/switch_controller]: DeserializationError cannot deserialize: unknown error handler name 'rosmsg'
[base_controller_spawner-2] process has died [pid 5395, exit code 1, cmd /opt/ros/kinetic/lib/controller_manager/spawner husky_velocity_controller husky_joint_publisher __name:=base_controller_spawner __log:=/home/administrator/.ros/log/0b5425c6-ec9a-11ea-83be-00e04c71ce5c/base_controller_spawner-2.log].
log file: /home/administrator/.ros/log/0b5425c6-ec9a-11ea-83be-00e04c71ce5c/base_controller_spawner-2*.log
`

civerachb-cpr, 2020-09-02:
It appears there was a bug in genpy that could cause this error.  The issue has since been resolved in genpy 0.6.13.

Can you please check if there are updates available on your system and if so, install them?

JackBorer, 2020-09-02:
I have 0.6.14 installed already.

civerachb-cpr, 2020-09-02:
Can you please share the contents of the log file mentioned at the end of your initial post?

JackBorer, 2020-09-02:
[rospy.client][INFO] 2020-09-01 17:33:05,515: init_node, name[/base_controller_spawner], pid[3079]
[xmlrpc][INFO] 2020-09-01 17:33:05,515: XML-RPC server binding to 0.0.0.0:0
[xmlrpc][INFO] 2020-09-01 17:33:05,515: Started XML-RPC server [http://cpr-aus05:35779/]
[rospy.init][INFO] 2020-09-01 17:33:05,515: ROS Slave URI: [http://cpr-aus05:35779/]
[rospy.impl.masterslave][INFO] 2020-09-01 17:33:05,516: _ready: http://cpr-aus05:35779/
[rospy.registration][INFO] 2020-09-01 17:33:05,516: Registering with master node http://localhost:11311
[xmlrpc][INFO] 2020-09-01 17:33:05,516: xml rpc node: starting XML-RPC server
[rospy.init][INFO] 2020-09-01 17:33:05,616: registered with master
[rospy.rosout][INFO] 2020-09-01 17:33:05,616: initializing /rosout core topic
[rospy.rosout][INFO] 2020-09-01 17:33:05,625: connected to core topic /rosout
[rospy.simtime][INFO] 2020-09-01 17:33:05,631: /use_sim_time is not set, will not subscribe to simulated time [/clock] topic
[rosout][INFO] 2020-09-01 17:33:05,644: Controller Spawner: Waiting for service controller_manager/load_controller
[rosout][INFO] 2020-09-01 17:33:05,651: Controller Spawner: Waiting for service controller_manager/switch_controller
[rosout][INFO] 2020-09-01 17:33:05,658: Controller Spawner: Waiting for service controller_manager/unload_controller
[rosout][INFO] 2020-09-01 17:33:05,665: Loading controller: husky_velocity_controller
[rospy.internal][INFO] 2020-09-01 17:33:05,801: topic[/rosout] adding connection to [/rosout], count 0
[rospy.msg][ERROR] 2020-09-01 17:33:05,807: cannot deserialize message: EXCEPTION Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/msg.py", line 223, in deserialize_messages
    msg_queue.append(data.deserialize(q))
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/controller_manager_msgs/srv/_LoadController.py", line 195, in deserialize
    codecs.lookup_error("rosmsg").msg_type = self._type
LookupError: unknown error handler name 'rosmsg'

[rospy.internal][ERROR] 2020-09-01 17:33:05,807: Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 728, in receive_once
    p.read_messages(b, msg_queue, sock)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 354, in read_messages
    rospy.msg.deserialize_messages(b, msg_queue, self.recv_data_class, queue_size=self.queue_size, max_msgs=1, start=1) #rospy.msg
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/msg.py", line 245, in deserialize_messages
    raise genpy.DeserializationError("cannot deserialize: %s"%str(e))
DeserializationError: cannot deserialize: unknown error handler name 'rosmsg'

[rospy.core][INFO] 2020-09-01 17:33:05,807: signal_shutdown [atexit]
[rosout][INFO] 2020-09-01 17:33:05,807: Shutting down spawner. Stopping and unloading controllers...
[rosout][INFO] 2020-09-01 17:33:05,807: Stopping all controllers...
[rospy.msg][ERROR] 2020-09-01 17:33:05,906: cannot deserialize message: EXCEPTION Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/msg.py", line 223, in deserialize_messages
    msg_queue.append(data.deserialize(q))
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/controller_manager_msgs/srv/_SwitchController.py", line 298, in deserialize
    codecs.lookup_error("rosmsg").msg_type = self._type
LookupError: unknown error handler name 'rosmsg'

[rospy.internal][ERROR] 2020-09-01 17:33:05,906: Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 728, in receive_once
    p.read_messages(b, msg_queue, sock)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 354, in read_messages
    rospy.msg.deserialize_messages(b, msg_queue, self.recv_data_class, queue_size=self.queue_size, max_msgs=1, start=1) #rospy.msg
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/msg.py", line 245, in deserialize_messages
    raise genpy.DeserializationError("cannot deserialize: %s"%str(e))
DeserializationError: cannot deserialize: unknown error handler name 'rosmsg'

[rosout][WARNING] 2020-09-01 17:33:05,906: Controller Spawner error while taking down controllers: transport error completing service call: receive_once[/controller_mana$
[rospy.internal][INFO] 2020-09-01 17:33:05,916: topic[/rosout] removing connection to /rosout
[rospy.impl.masterslave][INFO] 2020-09-01 17:33:05,916: atexit
[rospy.client][INFO] 2020-09-01 17:43:08,166: init_node, name[/base_controller_spawner], pid[5395]
[xmlrpc][INFO] 2020-09-01 17:43:08,166: XML-RPC server binding to 0.0.0.0:0
[xmlrpc][INFO] 2020-09-01 17:43:08,166: Started XML-RPC server [http://cpr-aus05:38859/]
[rospy.init][INFO] 2020-09-01 17:43:08,166: ROS Slave URI: [http://cpr-aus05:38859/]
[rospy.impl.masterslave][INFO] 2020-09-01 17:43:08,167: _ready: http://cpr-aus05:38859/
[rospy.registration][INFO] 2020-09-01 17:43:08,167: Registering with master node http://localhost:11311
[xmlrpc][INFO] 2020-09-01 17:43:08,167: xml rpc node: starting XML-RPC server
[rospy.init][INFO] 2020-09-01 17:43:08,267: registered with master
[rospy.rosout][INFO] 2020-09-01 17:43:08,267: initializing /rosout core topic
[rospy.rosout][INFO] 2020-09-01 17:43:08,276: connected to core topic /rosout
[rospy.simtime][INFO] 2020-09-01 17:43:08,278: /use_sim_time is not set, will not subscribe to simulated time [/clock] topic
[... cut ...]
    rospy.msg.deserialize_messages(b, msg_queue, self.recv_data_class, queue_size=self.queue_size, max_msgs=1, start=1) #rospy.msg
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/msg.py", line 245, in deserialize_messages
    raise genpy.DeserializationError("cannot deserialize: %s"%str(e))
DeserializationError: cannot deserialize: unknown error handler name 'rosmsg'

[rosout][WARNING] 2020-09-01 17:43:08,537: Controller Spawner error while taking down controllers: transport error completing service call: receive_once[/controller_mana$
[rospy.impl.masterslave][INFO] 2020-09-01 17:43:08,546: atexit

mikepurvis, 2020-09-02:
Please ensure that you've done an `apt dist-upgrade` so that you have the latest of all packages, including genmsg.

JackBorer, 2020-09-03:
As of Monday I had run apt dist-upgrade, all packages are fully up to date.

civerachb-cpr, 2020-09-03:
The issue may to be related to https://github.com/ros/genpy/issues/132.  Please let us know if the problem persists after re-checking for updates and installing them.

civerachb-cpr, 2020-09-03:
There was a near-identical issue reported for Ridgeback too, which was successfully resolved by installing the latest updates.
https://github.com/ridgeback/ridgeback_robot/issues/29

JackBorer, 2020-09-03:
I have run `sudo apt-get update` and `sudo apt-get dist-upgrade` and the problem still persists. My `genmsg` is version 0.5.11 which is the up to date version for Kinetic I guess? 

What confuses me is why on the kinetic-devel branch [https://github.com/ros/genmsg/commit/f9b1abe91e5c4742516c13124c9b8f7909aa98b2](url) it says it is 5.16.  But i guess its a core package so if updates don't absolutely have to be made they shouldn't?

But at this point I am no more close to the solution then before, with both genmsg and genpy up to date. Also I just want to say thank you guys so much for troubleshooting this!

OhmySallyGarden, 2020-09-04:
> It appears there was a bug in genpy that could cause this error. The issue has since been resolved in genpy 0.6.13.
> 
> Can you please check if there are updates available on your system and if so, install them?

This error doesn't show again after updating genpy. Thanks!

JackBorer, 2020-09-08:
I upgraded the Husky to 18 and Melodic and everything works as normal, and the controllers spawn as expected. I would have liked to figure out the original problem but deadlines cant wait :)

closed 2020-09-08 by JackBorer