husky/husky #147: Controller Spawner couldn't find the expected controller_manager ROS interface.

numb0824, 2020-10-23:
Hi,everyone,
     when i do:
    "roslaunch husky_gazebo husky_playpen.launch ",there are something wrong on the terminal:
   " waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
   Controller Spawner couldn't find the expected controller_manager ROS interface".
    i have tried my best to sove this problem,but didn't work.(i have already installed ros-kinetic-gazebo-control).
   could someone can help me?thanks

Carrot404, 2020-11-12:
Hi~
I have similar problems when i just install the package and "roslaunch husky_gazebo husky_empty_world.launch". The specific manifestation of the error is that the node gazebo is not linked to the velocity command node.
It bothers me a lot. If there is a solution , let me know please . Thanks a lot

civerachb-cpr, 2020-11-12:
A few questions:

- Are you using Melodic or Kinetic?
- Did you install husky_gazebo from-source or from the .deb package?
- Have you installed all available updates on your system?

Carrot404, 2020-11-13:
> A few questions:
> 
> * Are you using Melodic or Kinetic?
> * Did you install husky_gazebo from-source or from the .deb package?
> * Have you installed all available updates on your system?
hi
*Melodic
*from source i think . i cloned package from here.
*sorry im not sure about what you mean. 
Thanks in advance!

Carrot404, 2020-11-13:
@civerachb-cpr 
Thanks a lot. It is available after i update many packages.

civerachb-cpr, 2020-11-13:
Glad you got it working.

I'll leave this ticket open for now in case @numb0824 can provide more information.

illuSION-crypto, 2021-01-18:
I have the same problem. Unfortunately I tried a lot but useless. When I run "roslaunch husky_gazebo husky_empty_world.launch" it worked but when I run "roslaunch husky_gazebo husky_playpen.launch" it didn't. I'm using ros melodic on ubuntu 18.04. Neither from-source nor from-deb can work.

civerachb-cpr, 2021-01-18:
@illuSION-crypto are you seeing the same ROS output as the original poster?

```
waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
Controller Spawner couldn't find the expected controller_manager ROS interface
```

Or are you seeing something different?

illuSION-crypto, 2021-01-18:
> @illuSION-crypto are you seeing the same ROS output as the original poster?
> 
> ```
> waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
> Controller Spawner couldn't find the expected controller_manager ROS interface
> ```
> 
> Or are you seeing something different?

yes, just as you say

civerachb-cpr, 2021-01-18:
If you've installed from source, please make sure to run:

```
rosdep install --from-paths /path/to/src --ignore-src -r -y
```

Once that's done, or if you installed via apt and not from source, please run:

```
sudo apt-get update
sudo apt-get dist-upgrade
```

illuSION-crypto, 2021-01-18:
> If you've installed from source, please make sure to run:
> 
> ```
> rosdep install --from-paths /path/to/src --ignore-src -r -y
> ```
> 
> Once that's done, or if you installed via apt and not from source, please run:
> 
> ```
> sudo apt-get update
> sudo apt-get dist-upgrade
> ```

I tried both but still don't work. I check the log and find it is the same problem:
[rosout][INFO] 2021-01-19 00:07:00,063: Controller Spawner: Waiting for service controller_manager/load_controller
[rospy.internal][INFO] 2021-01-19 00:07:00,268: topic[/rosout] adding connection to [/rosout], count 0
[rospy.internal][INFO] 2021-01-19 00:07:02,177: topic[/clock] adding connection to [http://lenovo-ThinkPad-E575:33995/], count 0
[rosout][WARNING] 2021-01-19 00:07:30,307: Controller Spawner couldn't find the expected controller_manager ROS interface.
[rospy.core][INFO] 2021-01-19 00:07:30,311: signal_shutdown [atexit]
[rospy.internal][INFO] 2021-01-19 00:07:30,318: topic[/rosout] removing connection to /rosout
[rospy.internal][INFO] 2021-01-19 00:07:30,319: topic[/clock] removing connection to http://lenovo-ThinkPad-E575:33995/
[rospy.impl.masterslave][INFO] 2021-01-19 00:07:30,320: atexit
[rospy.internal][WARNING] 2021-01-19 00:07:30,321: Unknown error initiating TCP/IP socket to lenovo-ThinkPad-E575:56797 (http://lenovo-ThinkPad-E575:33995/): Traceback (most recent call last):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 563, in connect
    self.local_endpoint = self.socket.getsockname()
AttributeError: 'NoneType' object has no attribute 'getsockname'

civerachb-cpr, 2021-01-18:
It looks like the issue is actually somewhere deeper inside ROS and/or gazebo, not with Husky itself.  The fact that `self.socket` is apparent `None` is worrying.

Could you please try running

```
sudo apt-get install ros-melodic-gazebo-ros-control
```

illuSION-crypto, 2021-01-19:
> It looks like the issue is actually somewhere deeper inside ROS and/or gazebo, not with Husky itself. The fact that `self.socket` is apparent `None` is worrying.
> 
> Could you please try running
> 
> ```
> sudo apt-get install ros-melodic-gazebo-ros-control
> ```

Actually I already install it. Just now I found after I disconnect from the internet this problem disappear and there is a empty world in gazebo.
[image]
[image]
So I wonder whether models is being downloaded when ros running ?

illuSION-crypto, 2021-01-19:
OK I just solved this problem, when gazebo start it accesses a url which can't reach. I just accesses it using a http proxy

civerachb-cpr, 2021-01-19:
Glad you got it working!

closed 2021-01-19 by civerachb-cpr

chandailrc, 2021-03-09:
> > It looks like the issue is actually somewhere deeper inside ROS and/or gazebo, not with Husky itself. The fact that `self.socket` is apparent `None` is worrying.
> > Could you please try running
> > ```
> > sudo apt-get install ros-melodic-gazebo-ros-control
> > ```
> 
> OK I just solved this problem, when gazebo start it accesses a url which can't reach. I just accesses it using a http proxy. Anyway, really appreciate your help~

I am running into the same issue and not at all sure why. I have melodic and tried installing via. source as well as apt. I run into the same issue either way. Can you throw some light as to what the issue is and how to go about fixing it? I have updated the system packages (when installing via. apt) or performed rosdep dependency updates (when installing via. source) and it didn't seem to help. 

Kindly help!