husky/husky #148: C + + source code?

yufeilongyuan, 2020-11-17:
Why can't I find the C + + source code? Just started to learn, who can guide me, thank you。

civerachb-cpr, 2020-11-17:
This repository does contain the source code for Husky.  Most of the actual code is found in the husky_base package, inside the src folder.  The Husky firmware is closed-source, but can be installed on Linux by running `sudo apt-get install ros-$ROS_DISTRO-husky-firmware`.  If you're just working on a simulation in Gazebo the firmware won't even be necessary.

To compile the source code you need to install ROS on your computer.  Follow the instructions on the ROS wiki for that.  Then you should be able to run the following commands:

```
# first make a catkin workspace to check out the code:
mkdir -p ~/catkin_ws/src
cd ~/catkin_src
catkin_init_workspace src

# then clone this repository
cd src
git clone https://github.com/husky/husky

# install additional dependencies as-needed
cd ..
rosdep install --from-paths src --ignore-src -r -y

# finally compile the source code
catkin_make
```

There's very little actual code needed for Husky; most of it is in the husky_base package.  The rest is mostly launch files and configuration files for other ROS packages, which will have their own source code.

yufeilongyuan, 2020-11-18:
@civerachb-cpr Thank you very much for your reply, I probably know how to do it. However, there are still some minor issues to be consulted. For example, the navigation algorithm or control algorithm we are using, is the package of the ROS itself called? If I want to modify the navigation or control code, can I directly modify the ROS algorithm package? Compile later.....

civerachb-cpr, 2020-11-18:
I recommend going through a basic ROS tutorial, like this http://www.clearpathrobotics.com/assets/guides/melodic/ros/, or http://wiki.ros.org/ROS/Tutorials if you're not familiar with how ROS packages interact with each other.

Husky does not include any code for navigation.  It has code to support basic driving operation and sensor integration, but that's it.  But there are lots of 3rd party packages for navigation.  Gmapping is a relatively simple one for beginners, and the husky_navigation package is already set up to use it.  There's a guide on how to use the husky_navigation demos in the official docs here: http://www.clearpathrobotics.com/assets/guides/melodic/husky/

If you want to modify the navigation algorithms used, then you'll have to find the source code for e.g. Gmapping and modify that.

yufeilongyuan, 2020-11-19:
I'm very lucky to have your guidance. I already know what to do.

closed 2020-11-19 by yufeilongyuan