husky/husky #152: ROBOT_NETWORK parameter not setting ROS_IP

Gregory-Baker, 2020-12-04:
I have a Husky that I recently updated to Ubuntu 16.04 and Kinetic using the method outlined in the ROS - Husky tutorials. 

I was just now trying to configure the robot to network with my desktop PC, also running Kinetic. I added `export ROBOT_NETWORK=wlan0` to /etc/ros/setup.bash and I can see that this sets the (ROBOT_NETWORK) environment variable successfully, but nothing else seems to happen. Based on the [documentation](http://wiki.ros.org/husky_bringup/Tutorials/Customize%20Husky%20Configuration), I was expecting this to configure the ROS_IP variable, however this does not appear to be the case.

Things I've checked:
- The name of my wireless interface is definitely called `wlan0`

Any assistance would be much appreciated.

civerachb-cpr, 2020-12-23:
The ROBOT_NETWORK variable is passed to robot_upstart as one of the parameters for the Job class (see http://docs.ros.org/en/melodic/api/robot_upstart/html/jobs.html#robot_upstart.job.Job).  This in turn is used in the /usr/sbin/ros-start script that robot_upstart generates:

```
@[if interface]@
export ROS_IP=`rosrun robot_upstart getifip @(interface)`
if [ "$ROS_IP" = "" ]; then
  log err "@(name): No IP address on @(interface), cannot roslaunch."
  exit 1
fi
@[else]@
export ROS_HOSTNAME=$(hostname)
@[end if]@
```

(The above is taken from robot_upstart's job-start.em template, see https://github.com/clearpathrobotics/robot_upstart)

Could you please double-check the output of the ROS bringup service by running `sudo systemctl restart ros`, waiting a moment, and then running `sudo journalctl -u ros --since "1min ago"`?  Also, open up `/usr/sbin/ros-start` to make sure that the interface was correctly set when the file was created.

Gregory-Baker, 2021-09-28:
Thanks for the clarification!

closed 2021-09-28 by Gregory-Baker