husky/husky #154: [spawn_husky_model-11] process has died

illuSION-crypto, 2021-01-18:
I'm using ROS melodic on Ubuntu 18.04, when I tried to run "roslaunch husky_gazebo husky_playpen.launch" it came to this:
process[twist_mux-8]: started with pid [12240]
process[joy_teleop/joy_node-9]: started with pid [12245]
process[joy_teleop/teleop_twist_joy-10]: started with pid [12246]
process[spawn_husky_model-11]: started with pid [12259]
[ INFO] [1610949124.634990174]: [twist_marker_server] Initialized.
[ERROR] [1610949124.743112672]: Couldn't open joystick /dev/input/ps4. Will retry every second.
[ INFO] [1610949125.614141518]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1610949125.622572884]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[ INFO] [1610949125.873377953]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1610949125.880720975]: waitForService: Service [/gazebo_gui/set_physics_properties] has not been advertised, waiting...
[ INFO] [1610949127.609498685]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ WARN] [1610949127.837361231, 1298.480000000]: Failed to meet update rate! Took 1298.4800000000000182
[ WARN] [1610949127.842768658, 1298.480000000]: Failed to meet update rate! Took 1298.4600000000000364
[ERROR] [1610949127.843784, 1298.480000]: Spawn service failed. Exiting.
[ INFO] [1610949127.869738958, 1298.480000000]: Physics dynamic reconfigure ready.
Warning [parser_urdf.cc:1115] multiple inconsistent <gravity> exists due to fixed joint reduction overwriting previous value [true] with [false].
[spawn_husky_model-11] process has died [pid 12259, exit code 1, cmd /opt/ros/melodic/lib/gazebo_ros/spawn_model -x 0.0 -y 0.0 -z 0.0 -Y 0.0 -unpause -urdf -param robot_description -model / __name:=spawn_husky_model __log:=/home/lenovo/.ros/log/493e7c14-5951-11eb-a21f-5800e3fb5677/spawn_husky_model-11.log].
log file: /home/lenovo/.ros/log/493e7c14-5951-11eb-a21f-5800e3fb5677/spawn_husky_model-11*.log
Topic [///joint_cmd] is not valid.
Service [///joint_cmd_req] is not valid.
[ INFO] [1610949128.722222104, 1298.480000000]: Laser Plugin: Using the 'robotNamespace' param: '/'
[ INFO] [1610949128.722541211, 1298.480000000]: Starting Laser Plugin (ns = /)
[ INFO] [1610949128.725888586, 1298.480000000]: Laser Plugin (ns = /)  <tf_prefix_>, set to ""
[ INFO] [1610949128.842654002, 1298.480000000]: Loading gazebo_ros_control plugin
[ INFO] [1610949128.843214413, 1298.480000000]: Starting gazebo_ros_control plugin in namespace: /
[ INFO] [1610949128.844822519, 1298.480000000]: gazebo_ros_control plugin is waiting for model URDF in parameter [robot_description] on the ROS param server.
[ERROR] [1610949129.000507040, 1298.480000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/front_left_wheel
[ERROR] [1610949129.009983169, 1298.480000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/front_right_wheel
[ERROR] [1610949129.012222156, 1298.480000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/rear_left_wheel
[ERROR] [1610949129.013984752, 1298.480000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/rear_right_wheel
I can't find any issue or tutorial mentioned this problem, so I'm hoping help

closed 2021-01-18 by illuSION-crypto

civerachb-cpr, 2021-01-18:
I'm assuming that since you closed this issue you've found the solution yourself?

illuSION-crypto, 2021-01-18:
> I'm assuming that since you closed this issue you've found the solution yourself?

Unfortunately no. I closed it because I found it turned to the question "
Controller Spawner couldn't find the expected controller_manager ROS interface. "