husky/husky #155: How to enable/add some components in husky.urdf.xacro?

MuhammadUbuntu, 2021-01-18:
Greetings!
I need to add some components to husky robot such as UR5 manipulation and sensor_arch to husky.urdf.xacro.
Actually I have used husky_manipulation and husky packages and when I added the next line to husky.urdf.xacro:
` <xacro:include filename="$(find husky_ur_description)/urdf/husky_ur5_description.urdf.xacro" />`
the manipulation has been appeared with robot as following in gazebo:
` roslaunch husky_gazebo husky_empty_world.launch `
[image]

and to enable sensor_arch component, the husky.urdf.xacro already includes it in 
`  <xacro:include filename="$(find husky_description)/urdf/accessories/sensor_arch.urdf.xacro"/>`
and I have enabled it in `<xacro:arg name="realsense_enabled" default="true" />` but it doesn't appear in gazebo. Could you give me solution? and sorry for this dummy question.
Please note that I need to launch the robotic complex in gazebo.

civerachb-cpr, 2021-01-19:
I just ran

```
roslaunch husky_gazebo husky_empty_world.launch realsense_enabled:=true
```

and the sensor arch with RealSense D435 appeared as-expected.

Can you please share the output you're seeing in the terminal when you launch gazebo?

I see you've modified the base robot.  Can you please also share your customized URDF?

MuhammadUbuntu, 2021-01-21:
@civerachb-cpr Thank for reply. Actually, the problem was that value of `realsense_enabled `in launch file is overriding its value in urdf.xacro file i.e. I have made its value "true" in urdf file while its value in launch file was "false" so when I launched the file, it took the value from launch file and considered that  `realsense_enabled ` is false.
Could you please explain why I have to set ` HUSKY_URDF_EXTRAS` (and how) even though I have made the changes that I need.

civerachb-cpr, 2021-01-21:
`<xacro:arg name="realsense_enabled" default="false" />` appears multiple times across the launch files.  Depending _where_ you changed the default from `false` to `true` may be contributing to the problem.

If you only changed the default in `spawn_husky.launch` then the sensor bar will not appear when launching the empty world.  If you look at `husky_empty_world.launch` you'll see that launch file has its own `realsense_enabled` argument, which is passed directly to `spawn_husky.launch` which would override the default:

`husky_empty_world.launch`:
```
<arg name="laser_enabled" default="true"/>
<arg name="realsense_enabled" default="false"/>
[...]
<include file="$(find husky_gazebo)/launch/spawn_husky.launch">
    <arg name="laser_enabled" value="$(arg laser_enabled)"/>
    <arg name="realsense_enabled" value="$(arg realsense_enabled)"/>
</include>
```

You'll likely also want to set `laser_enabled` to `false` as otherwise a Sick LMS1xx will spawn on Husky's `front_mount` link, which you can see in both of our screenshots above (it's the black object clipping through the base joint of the arm).

> Could you please explain why I have to set HUSKY_URDF_EXTRAS (and how) even though I have made the changes that I need.

If you have already modified the Husky URDF as necessary you do not need to set the `HUSKY_URDF_EXTRAS` environment variable.  That variable is just a utility we provide to make customizing the Husky easier.  The way to set it is to add the line

```
export HUSKY_URDF_EXTRAS=/path/to/extras.urdf.xacro
```

to your bash setup.  On a physical robot this is normally done in `/etc/ros/setup.bash`.  On a desktop where you're working in Gazebo you could add it to `$HOME/.bashrc` or to `/opt/ros/melodic/setup.bash`.

This is optional, and if you've already modified the URDF file to add the arm (as it appears you have) then don't worry about the URDF extras variable.

closed 2021-01-21 by MuhammadUbuntu