husky/husky #157: Unable to launch Husky Gazebo simulator noetic-devel

EricWiener, 2021-01-25:
I started with a clean noetic environment:
```
mkdir -p ~/catkin_ws/src
$ cd ~/catkin_ws/src
$ sudo apt-get install ros-noetic-desktop-full
$ git clone https://github.com/clearpathrobotics/LMS1xx.git
$ git clone https://github.com/husky/husky.git
$ cd husky
$ git checkout noetic-devel
$ cd ~/catkin_ws
$ rosdep install --from-paths src --ignore-src --rosdistro ${ROS_DISTRO} -y --as-root pip:no
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
husky_bringup: Cannot locate rosdep definition for [um7]
husky_control: Cannot locate rosdep definition for [multimaster_launch]
husky_gazebo: Cannot locate rosdep definition for [multimaster_launch]
```

Wasn't sure how to solve that last issue, but I was successfully able to run:
```
catkin_make
source devel/setup.bash
```

But when I try to run Gazebo, I get the following error:
```
roslaunch husky_gazebo husky_empty_world.launch 
... logging to /root/.ros/log/6ae6a5e4-5e74-11eb-9a3b-0de85a64961c/roslaunch-cpr-mic09-31077.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://cpr-mic09:36295/

SUMMARY
========

PARAMETERS
 * /ekf_localization/base_link_frame: base_link
 * /ekf_localization/frequency: 50
 * /ekf_localization/imu0: imu/data
 * /ekf_localization/imu0_config: [False, False, Fa...
 * /ekf_localization/imu0_differential: True
 * /ekf_localization/imu0_queue_size: 10
 * /ekf_localization/imu0_remove_gravitational_acceleration: True
 * /ekf_localization/odom0: husky_velocity_co...
 * /ekf_localization/odom0_config: [False, False, Fa...
 * /ekf_localization/odom0_differential: False
 * /ekf_localization/odom0_queue_size: 10
 * /ekf_localization/odom_frame: odom
 * /ekf_localization/two_d_mode: True
 * /ekf_localization/world_frame: odom
 * /gazebo/enable_ros_network: True
 * /husky_joint_publisher/publish_rate: 50
 * /husky_joint_publisher/type: joint_state_contr...
 * /husky_velocity_controller/angular/z/has_acceleration_limits: True
 * /husky_velocity_controller/angular/z/has_velocity_limits: True
 * /husky_velocity_controller/angular/z/max_acceleration: 6.0
 * /husky_velocity_controller/angular/z/max_velocity: 2.0
 * /husky_velocity_controller/base_frame_id: base_link
 * /husky_velocity_controller/cmd_vel_timeout: 0.25
 * /husky_velocity_controller/enable_odom_tf: False
 * /husky_velocity_controller/estimate_velocity_from_position: False
 * /husky_velocity_controller/left_wheel: ['front_left_whee...
 * /husky_velocity_controller/linear/x/has_acceleration_limits: True
 * /husky_velocity_controller/linear/x/has_velocity_limits: True
 * /husky_velocity_controller/linear/x/max_acceleration: 3.0
 * /husky_velocity_controller/linear/x/max_velocity: 1.0
 * /husky_velocity_controller/pose_covariance_diagonal: [0.001, 0.001, 0....
 * /husky_velocity_controller/publish_rate: 50
[... 80 lines cut ...]
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/impl/tcpros_base.py", line 737, in receive_once
    self.stat_bytes += recv_buff(sock, b, p.buff_size)
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/impl/tcpros_base.py", line 109, in recv_buff
    raise TransportTerminated("unable to receive data from sender, check sender's logs for details")
rospy.exceptions.TransportTerminated: unable to receive data from sender, check sender's logs for details

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/ubuntu/catkin_ws/devel/lib/controller_manager/spawner", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/ubuntu/catkin_ws/src/ros_control/controller_manager/scripts/spawner", line 212, in <module>
    if __name__ == '__main__': main()
  File "/home/ubuntu/catkin_ws/src/ros_control/controller_manager/scripts/spawner", line 190, in main
    resp = load_controller(name)
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/impl/tcpros_service.py", line 442, in __call__
    return self.call(*args, **kwds)
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/impl/tcpros_service.py", line 532, in call
    raise ServiceException("transport error completing service call: %s"%(str(e)))
rospy.service.ServiceException: transport error completing service call: unable to receive data from sender, check sender's logs for details
[WARN] [1611536361.900845, 1.054000]: Controller Spawner error while taking down controllers: unable to connect to service: [Errno 111] Connection refused
[base_controller_spawner-3] process has died [pid 31095, exit code 1, cmd /home/ubuntu/catkin_ws/devel/lib/controller_manager/spawner husky_joint_publisher husky_velocity_controller __name:=base_controller_spawner __log:=/root/.ros/log/6ae6a5e4-5e74-11eb-9a3b-0de85a64961c/base_controller_spawner-3.log].
log file: /root/.ros/log/6ae6a5e4-5e74-11eb-9a3b-0de85a64961c/base_controller_spawner-3*.log
^C[spawn_husky_mo
```

civerachb-cpr, 2021-01-25:
The noetic branch is still being worked-on and hasn't been officially released yet.  Several of Husky's third-party dependencies have not yet been updated for Noetic.  And several files & scripts within the Husky repo still use older Kinetic/Melodic era formats (e.g. Python 2 scripts that need to be updated to Python 3).

EricWiener, 2021-01-25:
Got it. Thank you.

closed 2021-01-25 by EricWiener

Anjulo, 2021-07-12:
Hello,

So, when will the official noetic branch will be released?

Thank you.

civerachb-cpr, 2021-07-12:
We don't have an "official" release date targeted.  We're hoping to have all of our in-house testing on the physical robot done by the end of the summer.
For now, the husky_gazebo package in the noetic-devel branch is usable in Noetic if you need  to use the simulation environment (which is what this issue was originally asking about).