husky/husky #158: The Husky is not shown in gazebo

MuhammadUbuntu, 2021-01-27:
Greetings!
Today I tried to launch husky in gazebo as every time using `roslaunch husky_gazebo husky_empty_world.launch ` but husky isn't shown in gazebo.
Before I was able to launch it normally and I don't now what is the reason today. Anyway, the message that i have seen is: 

> IOError: [Errno 13] Permission denied: '/home/cont/.ros/rosdep/sources.cache/index'
> [spawn_husky_model-11] process has died [pid 16038, exit code 1, cmd /opt/ros/melodic/lib/gazebo_ros/spawn_model -x 0.0 -y 0.0 -z 0.0 -Y 0.0 -unpause -urdf -param robot_description -model / __name:=spawn_husky_model __log:=/home/cont/.ros/log/df739ff0-6083-11eb-8b60-00bb60079397/spawn_husky_model-11.log].
> log file: /home/cont/.ros/log/df739ff0-6083-11eb-8b60-00bb60079397/spawn_husky_model-11*.log
> [image]

civerachb-cpr, 2021-01-27:
It looks like when you configured `rosdep` you may have done so incorrectly; running `sudo rosdep update` can mangle the file permissions, resulting in these sorts of errors.  Are you able to launch any other ROS nodes on your system?  Or is it just husky_gazebo that's causing problems?  e.g. does running `roslaunch husky_viz view_model.launch` work as-expected?

You can try resetting them by running `sudo chown -R cont:cont /home/cont/.ros`. That may work.  Failing that, removing your `.ros` folder (or simply renaming it) and re-running `rosdep update` (without `sudo`) may get things back to a working state.

MuhammadUbuntu, 2021-01-28:
@civerachb-cpr Thank you, `sudo chown -R cont:cont /home/cont/.ros `fixed it directly.

closed 2021-01-28 by MuhammadUbuntu