husky/husky #159: Husky Frontier Exploration Demo: Unable to draw polygon to explore

EricWiener, 2021-01-30:
I'm following the navigation tutorials and everything is working great, but when I try to do the [Husky Frontier Exploration Demo](http://wiki.ros.org/husky_navigation/Tutorials/Husky%20Frontier%20Exploration%20Demo) tutorial, I'm unable to do this part:
```
Use the Point tool in the top toolbar to draw a closed polygon on the map that the Husky should explore. Watch the terminal window for instructions.
```

I'm able to set `2D Nav Goal`, but the simulated Husky doesn't take any actions when I click points. I also ran `rostopic echo /clicked_point` and validated my clicked points were showing up. Could I please have some advice on how to correct this.

I'm attaching an image of what this looks like below. I noticed that there is a warning that no exploration cost map has been received. 

[image]

Additionally, when looking through the logs, I get the message: 
```
[ WARN] [1612034369.700510976, 2008.840000000]: Timed out waiting for transform from base_link to map to become available before running costmap, tf error: canTransform: target_frame map does not exist.. canTransform returned after 2008.84 timeout was 0.1.
```
even though I checked that the transform exists.

[image]

Thank you!

civerachb-cpr, 2021-02-01:
Are you running the demo in Kinetic or Melodic?

EricWiener, 2021-02-01:
Melodic @civerachb-cpr

civerachb-cpr, 2021-02-02:
Looking at the screenshot, it looks like there's a warning saying no costmap was received.  That may be contributing to the issue, but might be a frontier_exploration issue, rather than a husky one.

Until very recently there wasn't a Melodic version of frontier_exploration available, and we haven't fully updated our melodic demos & docs to support it yet.  My advice would be to reach out to the frontier_exploration team directly for usage instructions, as I suspect it may be an error in our docs that we haven't had the opportunity to fix yet.  Sorry!

EricWiener, 2021-02-03:
Got it. I'll look into this further and update if I find any solution

closed 2021-02-03 by EricWiener