husky/husky #160: Is it possible to control wheels rotations directly?

MuhammadUbuntu, 2021-02-13:
Greetings!
I know that in husky it is used[ diff_drive_controller](http://wiki.ros.org/diff_drive_controller) from ROS to achieve the velocities control. Please could I know if it is possible to access to the low-level-control in Husky? That for example if I want to apply another controller rather than diff_drive_controller.

civerachb-cpr, 2021-02-16:
Because Husky is a differential drive robot, the diff_drive_controller is likely the best controller to use.  What controller do you want to use with Husky?

At a low-level, Husky communicates with the motors over a serial connection.  The horizon_legacy folder inside husky_base/src contains all of the low-level serial I/O, should you wish to modify any of it.

MuhammadUbuntu, 2021-02-17:
@civerachb-cpr  Thanks for reply.Does diif_drive_controller take in account uncertainties in mass? (for example if we need to load the robot, then the mass will change). 
if the answer is no, then I suggest to add adaptive algorithm. However, I am glad what you have said about diff_drive_controller.

civerachb-cpr, 2021-02-17:
The diff_drive_controller does not take the robot's mass into account, so if your robot needs constant acceleration with variable mass, this could theoretically cause problems.

The way Husky is configured, the diff_drive_controller is defined as the husky_velocity_controller in  control.yaml: https://github.com/husky/husky/blob/83a8a6ebef0b79f598222877ee8a1a238257ee49/husky_control/config/control.yaml#L5

You can override/replace that controller by either changing control.yaml, or creating a new yaml file and using the HUSKY_CONFIG_EXTRAS environment variable to point to it.  Values in the config_extras file will override anything loaded from control.yaml. You can see where the config files are loaded in config.launch:

1. Set the config_extras path: https://github.com/husky/husky/blob/83a8a6ebef0b79f598222877ee8a1a238257ee49/husky_control/launch/control.launch#L9
2. Load control.yaml: https://github.com/husky/husky/blob/83a8a6ebef0b79f598222877ee8a1a238257ee49/husky_control/launch/control.launch#L31
3. Load config extras: https://github.com/husky/husky/blob/83a8a6ebef0b79f598222877ee8a1a238257ee49/husky_control/launch/control.launch#L54  (This is the last line of the file, but occurs before any nodes are launched)
4. Spawn the controller manager & velocity controller, as defined in the config files loaded above: https://github.com/husky/husky/blob/83a8a6ebef0b79f598222877ee8a1a238257ee49/husky_control/launch/control.launch#L34

Whatever type your new controller is, you should make sure to name it "husky_velocity_controller" to ensure that the rest of Husky's nodes can communicate with it correctly.

MuhammadUbuntu, 2021-02-22:
@civerachb-cpr Could you please tell me how to set the control mode on "voltage control" instead of velocity control? It is needed in my low level controller to control the rotations of wheels.

closed 2021-03-20 by MuhammadUbuntu