husky/husky #161: husky gazebo issue

KeundongKang, 2021-03-04:
Hi,
I'm trying to launch husky gazebo in Ubuntu 20.04 but it is not working.
It just keeps loading like below

[image]
I definitely git cloned the noetic devel but it is not working.

Is there anyone know why?

Daegas, 2021-03-04:
What does the terminal sayy?

civerachb-cpr, 2021-03-04:
As @Daegas says, without seeing the terminal output it's pretty hard to diagnose the problem.

tonybaltovski, 2021-03-04:
Is this the first time you are you starting Gazebo on that machine? Gazebo needs to download models the first time you run it.

robogeekcanada, 2021-09-11:
@KeundongKang try to installing the following dependencies:

Reference: http://wiki.ros.org/LMS1xx
```$ sudo apt-get install ros-noetic-lms1xx```
Reference: https://bitbucket.org/DataspeedInc/velodyne_simulator/src/master/
```$ sudo apt-get install ros-noetic-velodyne-description```
Reference: http://wiki.ros.org/robot_localization
```$ sudo apt-get install ros-noetic-robot-localization```
Reference: http://wiki.ros.org/interactive_marker_twist_server
```$ sudo apt-get install ros-noetic-interactive-marker-twist-server```
Reference: http://wiki.ros.org/twist_mux
```$ sudo apt-get install ros-noetic-twist-mux```
Reference: http://wiki.ros.org/joy
```$ sudo apt-get install ros-noetic-joy```
Reference: http://wiki.ros.org/teleop_twist_joy
```$ sudo apt-get install ros-noetic-teleop-twist-joy```

tonybaltovski, 2021-11-09:
If installing from source use rosdep to install the dependencies. ie `rosdep install --from-paths src --ignore-src --rosdistro=$ROS_DISTRO -y`

closed 2022-03-16 by tonybaltovski