husky/husky #165: Logitech Controller under melodic distro?

RGring, 2021-03-15:
System Setup
 - ROS Distro: Melodic
 - provided ISO image installed: https://packages.clearpathrobotics.com/stable/images/latest/melodic-husky/amd64/
 - real hardware

I updated the real robot from kinetic to melodic. I followed the instructions here: https://www.clearpathrobotics.com/assets/guides/kinetic/husky/InstallHuskySoftware.html#installing-husky-software

ROS seems to work fine, but I can't control the robot with the Logitech controller (I exported the HUSKY_LOGITECH as stated in the instructions). Any help on that?

civerachb-cpr, 2021-03-15:
What button are you using as the deadman switch/enable button on the controller?  In Melodic we changed the enable button used on the F710 controllers.  Instead of the X button, you must now hold the LB button, located by your left index finger when holding the controller.  This change was made to keep the button mappings consistent across different robots and different controllers, as well as to make it easier to operate Husky with only one hand.

You can revert to the old behavior of you want by editing `husky_control/config/teleop-logitech.yaml` and changing the `button_enable` property.  There's a diagram in that file that shows the button and axis indices for the controller.

If you are holding down LB to enable the robot, the issue may be with the udev rule that handles the controller permissions.  With the USB dongle for the controller inserted into a USB port, can you please confirm that the Logitech input device is showing up in `/dev/input` by running

```
ls -l /dev/input
```

This should show a file called `/dev/input/f710`, which should be a symlink to `/dev/input/js0` (or similar, e.g. `js1`).

If that file doesn't exist, please ensure that `/etc/udev/rules.d/41-logitech.rules` exists and contains the following:

```
# Udev rule for the Logitech controller (e.g. the one shipped with Husky)
KERNEL=="js*", ATTRS{idVendor}=="046d", ATTRS{idProduct}=="c21f", SYMLINK="input/f710"
```

If that udev rule does exist already, and the controller device is still now being symlinked correctly, please let me know and I'll provide additional instructions.

RGring, 2021-03-15:
@civerachb-cpr Thanks for your reply. Indeed, I used the old button combination since I didn't stumble upon the information. That was an easy fix :). Thanks!

closed 2021-03-15 by RGring