husky/husky #167: Cannot control husky robot in multi_husky_playpen environment

CWEzio, 2021-03-18:
**Please provide the following information:**
 - OS: Ubuntu 18.04
 - ROS Distro: Melodic
 - Built from source or installed: built from source 
 - Package version: b552e16c6f2b0dd42e19e7cf116f51b97dbd3b0a
 - Real hardware or simulation: simulation

I run the multi_husky_playpen.launch and run 
```bash
rostopic pub -r 10 /sim/husky_gamma/husky_velocity_controller/cmd_vel geometry_msgs/Twist -- '[2.0,0.0,0.0]' '[0.0,0.0,1]'
```
 **Expected behaviour**
One husky should move 
 **Actual behaviour**
No husky moves and I got 
```bash
[ WARN] [1616059774.717023946, 1568.800000000]: ...not found
```
The whole output is 

```bash

... logging to /home/chenwang/.ros/log/d06676da-87cb-11eb-b611-c858c0d9f788/roslaunch-ROG-5760.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

Deprecated: xacro tag 'sick_lms1xx_mount' w/o 'xacro:' xml namespace prefix (will be forbidden in Noetic)
when processing file: /home/chenwang/catkin_ws/src/Husky_Pursuit/husky_description/urdf/husky.urdf.xacro
Use the following command to fix incorrect tag usage:
find . -iname "*.xacro" | xargs sed -i 's#<\([/]\?\)\(if\|unless\|include\|arg\|property\|macro\|insert_block\)#<\1xacro:\2#g'

Deprecated: xacro tag 'sick_lms1xx_mount' w/o 'xacro:' xml namespace prefix (will be forbidden in Noetic)
when processing file: /home/chenwang/catkin_ws/src/Husky_Pursuit/husky_description/urdf/husky.urdf.xacro
Use the following command to fix incorrect tag usage:
find . -iname "*.xacro" | xargs sed -i 's#<\([/]\?\)\(if\|unless\|include\|arg\|property\|macro\|insert_block\)#<\1xacro:\2#g'

Deprecated: xacro tag 'sick_lms1xx_mount' w/o 'xacro:' xml namespace prefix (will be forbidden in Noetic)
when processing file: /home/chenwang/catkin_ws/src/Husky_Pursuit/husky_description/urdf/husky.urdf.xacro
Use the following command to fix incorrect tag usage:
find . -iname "*.xacro" | xargs sed -i 's#<\([/]\?\)\(if\|unless\|include\|arg\|property\|macro\|insert_block\)#<\1xacro:\2#g'

started roslaunch server http://ROG:36219/

SUMMARY
========

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /gazebo_clock_relay/clock_relay_type: source
 * /gazebo_clock_relay/to: sim
 * /husky_alpha/from_sim/global_frames: ['map', 'world']
 * /husky_alpha/from_sim/prefix_operation: selective_remove
 * /husky_alpha/from_sim/services: [{'type': 'contro...
 * /husky_alpha/from_sim/tf_prefix: husky_alpha
 * /husky_alpha/from_sim/time_offset_operation: add_offset
 * /husky_alpha/from_sim/to: /sim/husky_alpha
 * /husky_alpha/from_sim/topics: [{'topic': 'joint...
 * /husky_alpha/husky_alpha_gazebo_export_relay/parameter_namespace: from_sim
 * /husky_alpha/husky_alpha_gazebo_import_relay/parameter_namespace: to_sim
 * /husky_alpha/husky_joint_publisher/publish_rate: 50
 * /husky_alpha/husky_joint_publisher/type: joint_state_contr...
 * /husky_alpha/husky_velocity_controller/angular/z/has_acceleration_limits: True
 * /husky_alpha/husky_velocity_controller/angular/z/has_velocity_limits: True
 * /husky_alpha/husky_velocity_controller/angular/z/max_acceleration: 6.0
 * /husky_alpha/husky_velocity_controller/angular/z/max_velocity: 2.0
 * /husky_alpha/husky_velocity_controller/base_frame_id: base_link
 * /husky_alpha/husky_velocity_controller/cmd_vel_timeout: 0.25
 * /husky_alpha/husky_velocity_controller/enable_odom_tf: False
 * /husky_alpha/husky_velocity_controller/estimate_velocity_from_position: False
[... 248 lines cut ...]
[ WARN] [1616059516.669351102, 1310.800000000]: ...not found
[ WARN] [1616059516.669477574, 1310.800000000]: ...not found
[ WARN] [1616059516.669683148, 1310.800000000]: ...not found
[ WARN] [1616059517.669809218, 1311.800000000]: ...not found
[ WARN] [1616059517.670008306, 1311.800000000]: ...not found
[ WARN] [1616059517.670209942, 1311.800000000]: ...not found
[ WARN] [1616059518.668987193, 1312.800000000]: ...not found
[ WARN] [1616059518.669121982, 1312.800000000]: ...not found
[ WARN] [1616059518.669305615, 1312.800000000]: ...not found

```

**To Reproduce**
Provide the steps to reproduce:
1.
```bash
roslaunch husky_gazebo multi_husky_playpen.launch
```
2. In a new terminal 
rostopic pub -r 10 /sim/husky_gamma/husky_velocity_controller/cmd_vel geometry_msgs/Twist -- '[2.0,0.0,0.0]' '[0.0,0.0,1]'

**Other notes**
I built multimaster tool from source. The code are downloaded from https://github.com/clearpathrobotics/cpr_multimaster_tools

Thank you very much for your help

CWEzio, 2021-03-18:
I suspect message_relay pacage goes wrong.

umnxslin, 2021-04-05:
I met with the same issue. But in the end I figured out there is no need to use multimaster even if you want to simulate multiple husky in gazebo. You can set **<arg name="multimaster" value="false"/>** in **multi_husky_playpen.launch**

Also you need to modify **spawn_husky.launch** by adding argument **<arg name="robot_namespace" value="$(arg robot_namespace)"/>** under <include file="$(find husky_control)/launch/control.launch">

In **control.launch**, add the argument **<arg name="robot_namespace" value="$(arg robot_namespace)"/>** under <include file="$(find husky_description)/launch/description.launch" >

And you are good to go. This works for me anyway

CWEzio, 2021-04-06:
@umnxslin Thank you. I have done exactly the same thing and it works for me too. I think that current multimaster has some bugs in it.

civerachb-cpr, 2021-04-07:
Thanks for bringing this to our attention.  We'll look into it, but I'm glad you were able to find a work-around for now.

closed 2024-07-24 by tonybaltovski