husky/husky #170: Slow simulation for multi husky in gazebo 9 ros melodic

Debojyoti7, 2021-04-08:
I'm running multiple huskies with a global planner in individual namespaces. However I find the simulation is slow. The gazebo 9 simulation is running with frequent freezing.  The exact same setup of husky package run perfectly without freezing in gazebo 7 in ros kinetic. 

Is there any way to solve this problem? 
My system specs is ubuntu 18. Ram 16gb

civerachb-cpr, 2021-04-19:
We've been seeing a lot of reports about issues with Gazebo 9, but so far haven't been able to nail down exactly what's causing the performance issues.

Are you able to run a single Husky simulation in Gazebo 9?  Or does the problem persist regardless of the number of robots?

For your system specs, what kind of GPU and CPU do you have in your computer?

Debojyoti7, 2021-04-19:
> We've been seeing a lot of reports about issues with Gazebo 9, but so far haven't been able to nail down exactly what's causing the performance issues.
>
> Are you able to run a single Husky simulation [...]

Hi  , my system has 16 gb ram, 1650 gpu