husky/husky #171: I'd like to understand why base_footprint is positioned below ground level

antoan, 2021-04-11:
https://github.com/husky/husky/blob/53959624e1a2fd6d2f0c84f103ef9d273e6f8e12/husky_description/urdf/husky.urdf.xacro#L106

which yields a negative z value of −0.13228

closed 2021-04-12 by antoan

civerachb-cpr, 2021-04-12:
I'm assuming since you closed this ticket you figured out the answer.  But just in case someone else finds this and is wondering the same thing:
The origin of the robot is at `base_link`, which is the center-bottom of the robot's main chassis.  The wheels extend below `base_link`, meaning the ground-plane has a negative-z value.  `base_footprint` is the outline of the robot projected onto the ground-plane.
This screenshot shows the `base_link` and `base_footprint`.  You can see that `base_footprint`'s z value lines up with the bottom of the robot's wheels.
[image]

mikepurvis, 2021-04-12:
... and the reason that base_link is attached to the chassis instead of the ground is that it's much more straightforward to measure the position of your sensors relative to that point. :)

antoan, 2021-04-12:
@mikepurvis @civerachb-cpr Much appreciated,  thank you.