husky/husky #172: How to get the feedback of Husky velocities?

MuhammadUbuntu, 2021-05-01:
Greetings!
We send velocities commands to Husky within cmd_vel. How we can get feedback of the measured velocities of Husky? I mean the real velocities of the robot and not we have sent to it.

civerachb-cpr, 2021-05-03:
You can get the angular velocity of each wheel from the `/joint_states` topic, which can be converted to the linear velocity by multiplying by the radius of the wheel.  On a standard Husky, the wheel radius is 0.1651m (see `husky_description/urdf/husky.urdf.xacro, line 78).  Obviously changes in tire pressure will affect the real-world radius, but this should give you a good approximation.

The `joint_states` topic will contain information like this:

```
header: 
  seq: 3953
  stamp: 
    secs: 1377
    nsecs: 860000000
  frame_id: ''
name: [front_left_wheel, front_right_wheel, rear_left_wheel, rear_right_wheel]
position: [19.085047243226207, 19.010108967194768, 13.544983137346293, 13.374554534481813]
velocity: [0.5672939631822598, 0.5672830724011997, 0.18267799254683934, 0.11022604243897091]
effort: [0.0, 0.0, 0.0, 0.0]
```

In the case above I am publishing the linear-x velocity of 0.1 m/s.  Multiplying 0.5672939631822598 rad/s by 0.1651 m gives 0.09366023332139109 m/s, which is within a rounding error of the published velocity.

Depending on the surface the robot is driving on there may also be external influences when it comes to converting the angular velocity to linear.  For example, loose sand or slippery surfaces may cause the robot's ground-speed to be lower than what the wheel velocities would indicate.

If you need to calculate the robot's speed relative to the environment more accurately than you can using the angular velocities of the wheels you may want to look into equipping the robot with a 2d lidar sensor and using some sort of SLAM system to calculate its speed using sensor data.

MuhammadUbuntu, 2021-05-06:
@civerachb-cpr Thank you for the detailed answer.

closed 2021-05-06 by MuhammadUbuntu