husky/husky #173: Customize teleoperation velocities

RGring, 2021-05-11:
Is there a convenient way to modify the teleop configuration on my husky? --> https://github.com/husky/husky/blob/melodic-devel/husky_control/config/teleop_logitech.yaml

I would like to reduce the rotational speed.

civerachb-cpr, 2021-05-11:
The easiest way would be to modify `husky_control/config/control.yaml` to edit the angular acceleration and angular velocity limits (see https://github.com/husky/husky/blob/34bd30535f11aad0092ad1217aa691621a29cc75/husky_control/config/control.yaml#L37).  This works best if you're working with a source build, but can also work if you're using the release version installed through a .deb.

If you're using the Husky packages installed through a .deb package, editing the file in /opt risks losing any customizations you've made when you update the package.  If this is a concern, you can create a config_extras file anywhere on your system and add this to `/etc/ros/setup.bash`:

```
export HUSKY_CONFIG_EXTRAS=/path/to/config_extras.yaml
```

That file should contain the following:

```
husky_velocity_controller:
  angular:
    z:
      max_velocity           : 1.5   # rad/s, default is 2.0
```

The `config_extras.yaml` file is loaded last, so it can be used to override any parameters loaded by `husky_control/launch/control.launch`.

RGring, 2021-05-12:
Hi @civerachb-cpr
Thanks for your detailed answer - HUSKY_CONFIG_EXTRAS is what I was looking for. I played a little bit around with the options.
* When I modifiy husky_velocity_controller/angular/z/max_velocity it doesn't affect the angular velocity of the teleop.
* It seems that I can't overwrite the joy_teleop via HUSKY_CONFIG_EXTRAS. Their values are still the default.
* Changing the config directly in /opt ... works, but is not the prettiest solution as you said.

civerachb-cpr, 2021-05-12:
A few possibilities:

1) Are you restarting ROS after setting the environment variable?  The parameters are loaded from the yaml files during startup, so modifying them after ROS has started won't have any effect until you run `sudo systemctl restart ros` (or otherwise restart the ROS nodes)

2) Are you exporting HUSKY_CONFIG_EXTRAS in `/etc/ros/setup.bash`, or are you setting it interactively in the terminal only?  It _must_ be set in /etc/ros/setup.bash for it to take effect; the ros systemd job runs as root (using setuidgid or setpriv to de-escalate its own privleges), so any envars you configure in .bashrc or interactively in the terminal are unknown to the systemd job.  /etc/ros/setup.bash is read by the systemd job to load any platform-specific envars needed for the ROS bringup

3) When changing the joy parameters, are you putting those parameters inside the `joy_teleop` namespace in the yaml file?  The teleop.yaml file included in husky_control omits this namespace because it is loaded within a group with that namespace inside the launch file.  HUSKY_CONFIG_EXTRAS is loaded outside any group namespaces, so you need to be more-explicit within the yaml file.

closed 2021-11-09 by tonybaltovski